supermerill/SuperSlicer #1426: Milling post processing disables raft function

well-engineered, 2021-07-18:
**Describe the bug**
On the E3D ToolChanger, I did set up an ASMBL-tool as Tool3, configured as a milling tool.
Now I can't turn on the raft at all. As soon as I change the raftlayers from 0 to 1 or more, the slicing fails.

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Set up a milling tool
2. try to enable raft

**>> Project File <<**
I can not export the project file, because of error shown in screenshot. As soon as there is no milling tool configured, export does work.

[image]

**Expected behavior**
"Enable raft" enables raft

**Desktop (please complete the following information):**

[image]

SuperSlicer
Version:   2.3.55
Build:     SuperSlicer_

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Version 10.16 (Build 20F71)
Total RAM size [MB]: 34,360MB
OpenGL installation
GL version:   2.1 ATI-4.5.14
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon Pro 560X OpenGL Engine
GLSL version: 1.20

neophyl, 2021-07-19:
Can you at least save the project without the raft enabled so that at least your machine and other profile settings are known.

supermerill, 2021-07-31:
can't reproduce, can you zip & post your projext here?

closed 2022-09-09 by supermerill