supermerill/SuperSlicer #1427: strange de-retractiob

Tinchus2009, 2021-07-19:
**Describe the bug**
Im having small blobs on my prints. Last test was tonight. I though this could be some problem related to stuttering, that problem was not happening before but still I checked it and it is not. After looking the sliced model again, I activated the "de-retraction" option and I think I found the problem because those de retraction are in the same place Im seeing these blobs.
The blobs are in this area

[image]

And in the slicer:

[image]

I have attached the project file. Version is 2.55.6, it is also happening in .7
[arm_plate.zip](https://github.com/supermerill/SuperSlicer/files/6840383/arm_plate.zip)

neophyl, 2021-07-19:
Just a general question, I see you have your printer set to use relative E distances, but I dont see any gcode in your startup to put the extruder into relative mode (G83).  Is that handled elsewhere ?

As far as I can see from the gcode output there are matching retract/unretract pairs.  Is it possible that your extruder is not fully retracting the specified amount and is pushing out the correct amount ?  That would give you a surplus at each/retract/unretract and its not unknown for bowden extruders to have those sort of problems.

Tinchus2009, 2021-07-19:
The firmware in the printer is already set to use relative extrusion, so no need to set it in the start gcode
Regarding the retractions: what code are you watching? because there are no retraction where the re retraction are made. These is the preview with tha retractions being shown:

[image]

And the same view with de retrations:

[image]

They are not maching in the area that is showing blobs

neophyl, 2021-07-19:
I’m not looking at the preview, I sliced the model and looked at the actual gcode produced (turned on verbose output). That will list all retract and unretract lines which if you search through the listing they all match. Each retraction is followed by an unretract.   Iirc there’s about 1944 of them in the output. 
The preview shows you the stuff on the outside but it’s harder to see something inside and they don’t always happen next to each other as it moves the start point on the next layer, depending on seam setting and model of course. 
Which is why I wondered if you have a difference in what your extruder is doing compared to what it’s being commanded to do.

Tinchus2009, 2021-07-19:
I dont understand: the preview is showing de-retratactions and no travel move. If I have a retraction, then a travel move should happend an a de-retraction. This is not what Im seeing here: in the picteure below in the certer I have focus on 2 de -retraction and I have activated the traves moves. As you can see there are 2 de-retraction, but no travel move there:

[image]

And just to be sure, If I move the nozzle, as you can see in the next picture, the path is being printed, then the retraction happens and then the path is being printed again, there is no jump to other place from the nozzle:

[image]

neophyl, 2021-07-20:
One thing you have to realise about the preview is that the blue unretractions can obscure the purple retractions.  Look at layer 0.2 and follow the nozzle with just the retractions enabled and note where they are placed.  Now turn on untractions too and repeat.  You will see lots of blue dots and not so many of the purple ones that you could see before.  It

As I say, look in the actual gcode produced.  Your model sliced has this as an example for that layer (at around line 1282 when I sliced it) 

G1 E-4.50000 F1500 ; retract
G1 Z1.200 F18000 ; lift Z
G1 X205.221 Y119.260 ; move to first Internal perimeter point
G1 Z0.200 ; restore layer Z
G1 E4.50000 F1500 ; unretract

as you can see there is a retract, a lift, move to the internal perimeter point, a lower and then a unretract.  All as I would expect given your settings.  Each of the retract commands has a following unretract command (well all except the last one at line 408633) but that's not unexpected as it goes to the end gcode at that point.

As you have retract at layer change enabled then you wont always have a travel move as if it starts a new layer at the same spot it ends the last layer then it will do a retract/unretract every time theres a z lift.  Also has you have a minimum travel of 1mm then the whole retraction process isnt done for moves less than this distance.

And that about gets to the limits of my knowledge so anything further will have to be by someone else.  I hope you get your issue resolved as I cant see anything wrong with the gcode output when it comes to retract/unretract.

Tinchus2009, 2021-07-20:
Nephyl, we are stuck, you are just looking at te spots were retraction are ok. In the last image I already did this test and posted the images at the beginning: retraction are not in the same place of de retraction. And the last image talks by itself: the nozzle makes a path and suddenly there is a de retraction with any junp, layer change, previous retraction.
And you are ignoring too the travelling: there are no travel moves going IN or OUT of those unretractions.

supermerill, 2021-07-20:
The ghosty deretraction are a combination of insufficient `xy gcode decimals` that hide a problem with `minimum extrusion length` (it can create long- veryshort extrusions, will fix that)

closed 2021-09-06 by supermerill

LAP87, 2022-06-06:
I'm seeing this phenomena in the 2.4.58.2 build, i think.. lots of unrelated deretractions being done on external perimeter for no obvious reason, leaving bumps of extra material that end up visible on the finished print.
I've been testing all kinds of settings and reslicing it over and over, the spots never move or disappear.
[image]