supermerill/SuperSlicer #1432: (unsure) Overhang Perimeter instead of Bridging

ForsakenHarmony, 2021-07-20:
**Describe the bug**
The slicing on this model seems problematic, the outside is overhang perimeter (which is fine, but weird) and the inside is added without any support, which I don't want to add because it would defeat the point of the model.

This leads to the printer extruding into nothing, because it has no chance to stick.

**To Reproduce**
Slice [PolyPearl Tower](https://www.thingiverse.com/thing:2064029)

**>> Project File <<**
[PolyPearl_Tower.zip](https://github.com/supermerill/SuperSlicer/files/6851332/PolyPearl_Tower.zip)

**Expected behavior**
I expect there to be normal bridging in this case, though the overhang might be hard to fix.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS:  Windows
 - Version: 2.3.56.7

supermerill, 2021-07-21:
I think you can try that:
[PolyPearl_Tower_bitbetter.zip](https://github.com/supermerill/SuperSlicer/files/6856265/PolyPearl_Tower_bitbetter.zip)

the first modifier try a perimeter-only strategy, and the other one the max-bridge you maybe asked for.

ForsakenHarmony, 2021-08-16:
Yeah, that's better, still not ideal, but not sure what the ideal solution to slice this without support would be

closed 2021-08-16 by ForsakenHarmony