supermerill/SuperSlicer #1434: Strange stuttering sound on inner brim perimiter

mrvanes, 2021-07-21:
**Describe the bug**
I have most of the Slicer parameters dialed in, but one weird effect remains: the last (inner) brim perimeter causes a fast stuttering, humming vibration while printing. None of the other layers or features suffer this effect. I've reduced the max_accel and square_corner_velocity in Klipper, no relief.

**To Reproduce**
Steps to reproduce the behavior:
1. Have my printer (or not?)
2. Load my project
3. Print
4. Listen to the printer while printing the inner brim perimeter

**Project File**
Hope this contains everything needed?
This file is not zipped, only renamed since Linux refuses to zip a zipfile.
[20140205_Marvin_KeyChain.3mf.zip](https://github.com/supermerill/SuperSlicer/files/6855833/20140205_Marvin_KeyChain.3mf.zip)

**Expected behavior**
No vibration/stuttering while printing

**Screenshot**
[image]

**Desktop**
 - OS: Linux
 - Version 2.3.56.7

supermerill, 2021-07-21:
can you try with:
* latest nightly https://github.com/supermerill/SuperSlicer/suites/3291315685/artifacts/76849456
* 0.35 as "min length" (in the printer general setting)
?

mrvanes, 2021-07-21:
Thx. It seems the 0.35 min length solved most, if not all, of the problem.

closed 2021-07-21 by mrvanes

mrvanes, 2021-07-23:
It turns out there is also strange interaction between Klipper's acceleration limiting. I let Klipper do all the limiting (disabled SuperSlicer limits) but this seems to cause the erratic behaviour. I now set SuperSlicer to safeguard and configured all Klipper's limits higher so that they effectively don't interfere with SuperSlicer's limit. This seems to work the best for my printer?

supermerill, 2021-07-23:
> and configured all Klipper's limits higher so that they effectively don't interfere with SuperSlicer's limit

This shouldn't be necessary. If you are using the limits as safeguards, they should be the same as what you entered in your firmware. If the gcode output something bigger than you have set, then there is a bug.