supermerill/SuperSlicer #1435: web access slicer

tjengbudi, 2021-07-22:
is there any chance to make this slicer can access by web? it is very convenient to access this profile event only local network. it would really nice feature

supermerill, 2021-07-22:
Do you means cloud-based profiles?
Or webapp that run into a browser?

tjengbudi, 2021-07-22:
not the 2 of them
it will really nice to have some feature like utorrent. if the web management is activated. i can access them by web with local network setting. it means i just open with ip address that program running and the port.

i believe delphi have component to do that. but i do not know how it works. cause if i remember it is paid component. i do not know either how to implement in c

now i just used remote desktop to do that. but i hope it will have some feature like this. it will really nice

supermerill, 2021-07-22:
I don't understand

tjengbudi, 2021-07-22:
sorry for my bad explanation

here is screen capture where utorrent have this feature
[image]

supermerill, 2021-07-22:
what feature? i don't understand a word. Please explain to me like if i was a 5 year old.

tjengbudi, 2021-07-23:
the web ui is the way to control interface from web browser. it is replicate the main control just like it is control the main apps. the behaviour is the same. just activate the web ui setting. access by web browser ( for example if the port is 8080) url 127.0.0.1:8080 and you can access it. it is the best way to have 1 maintain profile and share connectivity in local network. 

sorry my english might not good to explain. i try my best. thx u

supermerill, 2021-07-23:
Ok
I won't do that, it's years of man-hour of dev.

tjengbudi, 2021-07-23:
noted

closed 2021-07-23 by tjengbudi