supermerill/SuperSlicer #1436: Number of layers affected by XYcompensation->Holes and XYcompensation->First layer

laptop-laptop, 2021-07-25:
**Is your feature request related to a problem? Please describe.**

Hi,

For XYcompensation->Holes and XYcompensation->First layer, how many layers do they affect? After spending almost 10 hours on printing, I found that changing these values not only made the holes at the bottom larger also enlarged the holes on top layer. Isn't this a bug?

**Describe the solution you'd like**
Allow user to determine how many layers and which layers are affected

**Describe how it would work**
For example, if I set only layers 1 to 3 are affected, the hole size in only layers 1 to 3 are enlarged. The rest remain the same as in the original STL file.

supermerill, 2021-07-25:
version?

First layer affect the first layer

holes affect all layers

laptop-laptop, 2021-08-04:
2.2.53

By ""Holes" affect all layers with holes"", do you mean only vertical holes or vertical holes and also inclined holes (e.g. holes that are at an angle from the vertical axis)?

supermerill, 2021-08-05:
It works on the layer polygon, so it will modify the layer if the detector recognize a holes with good properties inside the layer.

closed 2021-09-06 by supermerill