supermerill/SuperSlicer #1439: Strange bug with brim cause enabling "External perimeter first"

Pavulon87, 2021-07-27:
**Describe the bug**
Same project is creating different gcode after turning brim on/off and reloading

**To Reproduce**
Steps to reproduce the behavior:
1. Open attached project
2. Enable/disable brim
3. Slice
4. Save project
5. Reopen project
6. Slice again

**>> Project File <<**
[outer_first.3mf.zip](https://github.com/supermerill/SuperSlicer/files/6883745/outer_first.3mf.zip)

[inner_first.gcode.txt](https://github.com/supermerill/SuperSlicer/files/6883785/inner_first.gcode.txt)
[outer_first.gcode.txt](https://github.com/supermerill/SuperSlicer/files/6883786/outer_first.gcode.txt)

**Expected behavior**
 "External perimeter first" should not be affected by brim or other options

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 2.3.56.7

supermerill, 2021-07-27:
If you have a brim, the first layer should be with an External perimeter first

supermerill, 2021-07-31:
can't reproduce with your procedure.

closed 2021-07-31 by Pavulon87