supermerill/SuperSlicer #1440: Small movements on external perimeters

randellhodges, 2021-07-28:
**Describe the bug**
External perimeters movements broken up into multiple 0.10mm moves around 90 degree corners. Doesn't happen with internal perimeters. I'm not sure if this is by design, a configuration option, or a bug.

**To Reproduce**
Steps to reproduce the behavior:
1. Slice
2. View G-Code
3. Layer: 143

**>> Project File <<**
[Body10.zip](https://github.com/supermerill/SuperSlicer/files/6889959/Body10.zip)

**Expected behavior**
The outer movements shouldn't be broken up and should move like the internal perimeters.

**Screenshots**
Internal Perimeters (working like I expect)
[image]

External Perimeters (note the 2 small movements of 0.10mm before it finishes out the move)
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 2.3.56.7

supermerill, 2021-07-28:
If you don't want the `cutting corners` feature, deactivate it

randellhodges, 2021-07-28:
> If you don't want the `cutting corners` feature, deactivate it

Thank you. I can confirm changing this makes it work as I was expecting. Not sure when I cranked that to 100%.

closed 2021-07-28 by randellhodges