supermerill/SuperSlicer #1441: Ability to have different layer heights with each imported part.

terryrobison, 2021-07-28:
Would it be possible to implement different layer heights with different parts in a project?  

This would be useful for multi-nozzle / tool changing machines where thick layers could be used in a specified area.  Specifically, inserting a nylon bush within a functional part.  A large diameter nozzle could be selected for this part printing the internal bush, say 0.6 mm layer height.  A more typical nozzle diameter could be used on the other part too encase the strong thick bushing.  The 0.4 mm nozzle would provide a more pleasing outer shell of the functional part where cosmetic considerations are important.
Right-clicking on the second part and setting the print height of this part overriding the global print height could be a way of doing this.  

I believe Simplify3D can print different layer heights for different objects on the build plate via “processes”.  I don’t have this slicer, so I can’t comment on that. 

I have four print nozzles and would like to use my 0.8mm tool head for thick strong internal sections of a part, and 0.4mm tool head for finer details in other sections of the print.  Layer heights seem to be a global setting in SuperSlicer.   It would be fantastic to have this set per object imported into a job.

neophyl, 2021-07-28:
This is already possible.  Set up multiple extruders and then assign the different extruders to the different parts.

Other settings can be applied to different items/areas by using modifier meshes or layer height modifiers.

supermerill, 2021-07-28:
It's a bit confusing.

You ask for `different layer heights` (0.2mm perimeter and 0.6mm infill ?)
But the reason is using different nozzles (which is already possible, like neophyl said).

It's problematic to use different layer height for features because when there is a slope, the infill section can be buried below the perimeters.

terryrobison, 2021-07-30:
I have multiple extruders (E3D Toolchanger) with nozzles ranging from 0.4 - 0.8mm.  I can load a part and set the layer height.  When I add another part, I can't locate where I can change the layer height on the second part.  I can locate perimeters, infill etc but no layer thickness.  I can only change this in a global context in the print settings.  I'm sure it must be there, I have failed to locate it.
Regards,

neophyl, 2021-07-30:
Select your 'other' object.  Right click and select perimeter and shell.  That adds 4 settings to the right hand panel for that object.  Bottom solid layers, BASE LAYER HEIGHT (the one you want), Perimeters count and Top Solid Layers.
I just tried this with 2.3.56 and also with 2.2.54 so the option has been in there awhile.  Its one way of printing different objects at different layer heights even with a single extruder without using the variable layer height function.

terryrobison, 2021-07-31:
I have Version 2.3.56.7.   The view is set to Expert.  I am missing the Layer height, only three options are available.  I was looking in the correct place, but it's not exposed in the available options.   Many thanks. 
[image]

neophyl, 2021-07-31:
No idea off hand why that setting isn’t showing for you. Could you please save your project file , zip it up and attach it.  Maybe with the full project we can figure out why you don’t have it.

terryrobison, 2021-08-02:
I think I figured it out.  If I click File -> Import -> Import STL.  Then right-click on the part in the 3D view, and select ->Add Part, to import a second part.  Then Menu selection for height is not there.  
If I click File -> Import -> Import STL, then repeat the same steps of File -> Import -> Import STL (now I have two parts in the 3D view) the Menu for layer height is there.   
Driver error on my part?
Regards,

supermerill, 2021-08-02:
If you add a second object via right-clic->add part, it add a sub-part to an existing one, to be able to do a multi-material print.
That way, the two stl are linked to the same origin and they are considered as only on object.

If you go via the menu of the `Add...` widget, then you add a new independent object into the plater

terryrobison, 2021-08-02:
Cheers, thank you so much for explaining that.  It makes sense now.  Once again. thank you for your help.

closed 2021-09-06 by supermerill