supermerill/SuperSlicer #1444: Over-extrusion

mafotaso, 2021-07-30:
**Describe the bug**
When you laminate, show perimeter around 0.95mm width and I have selected 0.4nozzle and 0.44mm width max. That cause me a lot  of printing obstruction and fail. 

The problem  is the space, slicer doesn't understand that it can print 2 lines and not only 1 with a big width. I have tested it in differents versions, differents size of the piece and nothing work. The onyl thing that you can to do is decrease perimetral width and increase external perimetal and you can improve it a little. 

**To Reproduce**
1. Go to laminate
2. Click on 
3. Scroll down to width view
4. See error around last layers near of the ears.

**>> Project File <<**
[catplanter.zip](https://github.com/supermerill/SuperSlicer/files/6909300/catplanter.zip)

**Expected behavior**
just lamine

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 20H2 19042.1110
 - Version 2.3.56.7

graflash, 2021-07-30:
Enable "Also for all perimeters"
[image]
[image]

mafotaso, 2021-07-31:
you are a genius!!! 

solved! 
[image]

closed 2021-07-31 by supermerill