supermerill/SuperSlicer #1445: Is there a simple tool to translate the language?

SphaeroX, 2021-07-31:
I have already read in the wiki that there is a possibility for translations (https://github.com/supermerill/SuperSlicer/wiki/Creating---updating-a-translation).

Unfortunately, this does not seem to be up to date, because there is still worked with java but currently everything about python goes?

So I looked at the .pot and quickly found that in the German translation there is already the text, but just with a different comment, here is an example:

_**localization/de/slicer.pot**_

> #: src/libslic3r/PrintConfig.cpp:984
> msgid "Position of perimeters starting points."
> msgstr "Position des Startpunktes des Umfangs."

_**localization/slic3r.pot**_

> #: src/libslic3r/PrintConfig.cpp:3177
> msgid ""
> "Position of perimeters' starting points.\n"
> " "
> msgstr ""

But this text is not translated in the current SuperSlicer version. 

I would like to help and I think many others too but the hurdles are unfortunately too hard if you are not in the matter. Couldn't you write a small tool so that everyone can help?

So simply the sentence is displayed in English and then you can type it in your selected language. At the end the file comes out completely ready and you can put it here.

supermerill, 2021-07-31:
yes, it was moved into https://github.com/supermerill/SuperSlicer/tree/master/resources/localization

i'll update the wiki

SphaeroX, 2021-08-02:
Okay i try it but on windows the msgunfmt commadn wont works... Maybe im to stupid for that.

Other Question, cant someone makes an simple YouTube Video for Windows?

supermerill, 2021-08-02:
> Okay i try it but on windows the msgunfmt commadn wont works... Maybe im to stupid for that.

If you have some problems with it, you can download the (most recent) binaries [here](https://github.com/mlocati/gettext-iconv-windows/releases) or the outdated [here](http://gnuwin32.sourceforge.net/packages/gettext.htm) (use the installer or be sure they are in a directory that is in your PATH, ie it's called from a command line)

> Other Question, cant someone makes an simple YouTube Video for Windows?

I can, but that takes time, and can be easily obsolete if something changed.

5axes, 2021-08-03:
I'm personnaly I'm using the "old" ou previous solution to generate the SuperSlicer.po and SuperSlicer.mo file.  In the new Ptython script used call  "pom_merger.py" we need to have the python-Levenshtein module installed but according to my research the 
python-Levenshtein has something to do with a C library or something like that .. The basic previous script was much more simple to use because you just need to use Python and no other additionnal library. Perhap's you could use the old script.

supermerill, 2021-08-05:
> we need to have the python-Levenshtein module installed but according to my research the python-Levenshtein has something to do with a C library or something like that ..

You can't install it with a pip install python-Levenshtein ?
I may add a switch to disable the little part that use it if not installed  (if it can't be installed on some computers).

5axes, 2021-08-05:
> You can't install it with a pip install python-Levenshtein ?

No I'm sorry I don't remember exactly the error message but after an hour , tryong to install the python-Levenshtein module, has the previous script was working fine , as I was lazy I prefered to stay on the old script.

closed 2023-11-27 by SphaeroX