supermerill/SuperSlicer #1446: Option to disable collision warnings for sequential printing

olsonmax, 2021-07-31:
I've been playing with different ways to embed 1-3 layer thick inlays in an accent colour into the bottom surfaces of larger print objects. While there are techniques involving slicing two gcode files (one for the accent body/bodies, and one for the main body), and executing them back to back without allowing a cooldown, it would be great to have the ability to execute 2-material, 1-nozzle prints from a single gcode file with only one filament swap. 

Of course, this would require the smaller internal inlay object(s) to be printed first, be at most a few layers high, and have a generous inward chamfer on the sides, with the main body to be printed last having an enforced z-hop of sufficient height to avoid collision with the existing inlay object(s).

Unfortunately, the collision detection system for sequential printing doesn't allow for two objects to be within each others' borders. Adjusting the options for sequential printing like extruder clearance and single-skirt enforcement doesn't get around this.

**Describe the solution you'd like**
Ideally, a way to disable the `sequential_print_horizontal_clearance_valid` and `sequential_print_vertical_clearance_valid` checks that prevent slicing.

**Additional context**
The screenshots show an example of what I'm trying to achieve:

### Cross-section
[image]
### All objects in place
[image]
### Separated for clarity
[image]
[image]

supermerill, 2021-07-31:
You do that because you only want to do 1 manual swap vs 3 manual swap, right?

If so, please note that to avoid collision, you need an `external perimeter width` > nozzle **external** diameter

I'll see to deactivate the check if it's set to 0

olsonmax, 2021-07-31:
> You do that because you only want to do 1 manual swap vs 3 manual swap, right?
>
Yes, exactly. 1 vs 3 manual swaps for a two layer inlay, or 1 vs 5 for a three layer inlay.

> If so, please note that to avoid collision, you need an `external perimeter width` > nozzle **external** diameter (for the first 3 layers)
> 
A good tip, and something I hadn't considered doing. I was hoping I could get away with some "plowing" (especially since the closeness to the bed should keep the existing plastic somewhat soft) but this is much better.

> I'll see to deactivate the check if it's set to 0
>
That would be perfect. Do you mean to disable
1. the vertical check if vertical clearance is set to 0, **and**
2. the horizontal check if horizontal clearance is set to 0?

I was encountering both errors as I was trying to figure it out.

Thanks for all your hard work!

supermerill, 2021-08-01:
For the vertical check, just put a high enough number in the adequate setting, something like 9999.