supermerill/SuperSlicer #1449: Thin layer slicing.

awtew, 2021-08-02:
### Version
2.3.56.6

### Operating system type + version
Window 10

### Behavior
Slicer does not slice a thin object z < 1 mm correctly.
Attached please find two STLs mesh files. Both have a base and a thin 0,2 mm edge on top.
[example files.zip](https://github.com/supermerill/SuperSlicer/files/6915898/example.files.zip)
Component1.stl was exported from Fusion 360
torquewrench-depthtest.stl was exported from Freecad.
The results are the same.

The slicer calculates the toolhead for the thin top layer but does not calculate any toolpaths for the base. Both have a .2 mm edge but have different depths to differentiate.
It may be a setting but other slicers are slicing this correctly (CURA, simplify 3d). Prusa Slicer shows the same issue.
I may have missed a setting but the behavior is strange.

supermerill, 2021-08-02:
I don't see the problem
[image]

awtew, 2021-08-02:
This is what the GCODE looks like a few steps before the end
It never cacs the bottom at all only the ledge.
[image]

[image]
[image]
and then the image you get at the end. 

and then the last slide shows the completed finish as per your image.

supermerill, 2021-08-02:
yes, i don't understand the issue.

> It never cacs the bottom at all only the ledge.

what "cacs" and "ledge" means? I don't understand the sentence.

neophyl, 2021-08-02:
I'm a native English speaker and I can't follow it either Merill.  I can also slice the 2 included stl files correctly, the gcode is generated properly and the preview shows all the layers correctly too.  Then again as the OP hasn't included a saved 3mf project file we cant tell if any of his settings are causing an issue. 

I also note the screen captures provided just show the nozzle movement on a single layer and not the layer slider view so I have no idea what the actual issue is.  Perhaps they are confused how the nozzle preview works ?  In that it only shows that layer and when it gets to the end of the layer it then shows the completed model.  The nozzle preview is per layer and the slider on the right is to select the layer you need the more detailed preview of.  That horizontal slider is not for showing the nozzle movements from the start of the print to the end of the print.  Well not by default anyway.  If you want the sequential slider (the horizontal slider) to apply to all layers then you need to go into Configuration>Preferences, select the Gui tab and then untick Sequential slider applied onto to top layer.

btw Ledge is a perfectly valid English word, its different from edge.  "a narrow horizontal surface projecting from a wall, cliff, or other surface." or "an underwater ridge, especially of rocks beneath the sea near the shore".  Although I'm struggling to see how this word relates to the posted models as neither has a Ledge in the design.

supermerill, 2021-08-02:
thx
TIL

qwewer0, 2021-08-02:
@awtew What is your "Z full step"?

awtew, 2021-08-03:
Dear all,

Thanks for your help. @neophyl you are correct it was an edge on the top not a ledge. Your suggestion on the configuration was indeed what I needed thanks for that. 
I am happy to report that we can close this issue.

closed 2021-08-03 by awtew