supermerill/SuperSlicer #1451: Multiple Partcooling Fan Support?

nicedevil007, 2021-08-03:
**Is your feature request related to a problem? Please describe.**
I have 2 Fan ports on my Biqu BX that I want to use. I can manually enable the 2nd fan by switching on the Display UI the fanport and enable it there.

**Describe the solution you'd like**
Having a button to setup the number of part cooling fan ports that are in use.

**Describe how it would work**
GCODE addition for part cooling like so?
M106 P0 S255 and M106 P1 S255 should do the job then f.e. with 2 fans.

supermerill, 2021-08-03:
Why not using both at the same time?
It's more of a firmware job.
I can't see what I can do with a second fan.

nicedevil007, 2021-08-04:
heya supermerill (everytime I read your name I got remembered of Lord of the Rings... dont know why, sounds like a name of an actor :) ), back to topic:

You are absolutly right, in most cases (I would say 99%) you would have the same speed on both fan ports. In my case I have a Biqu BX that got 2 ports for fans at the daughterboard sitting on the backside of the extruder. 
1 Fan is not enough to get good cooling done here, so I installed a 2nd one at the front. Then I was able to enable 2nd fan manually with the gui of the display but not inside the slicer.

I know that I could solder both fans together to 1 cable and just use one port but then I asked myself, why not using both if both are provided my default.

Then I figured out that the slicer doesn't support mutiple fans. Thats why I was asking here for a small option to implement. Default should be only 1 fan port but if we have multiple ones than just tell the slicer how much fanports it should tell to run a fan and it is doing the gcode above (with right speed ofc). Different speeds aren't needed in my opinion.

Some guys that got the same issue were trying to use a postprocessing script to get rid of this problem, but why not asking the dev to make something that is still not included in prusaslicer :)

supermerill, 2021-08-05:
Even if I output the gcode you suggested, you will need to modify the firmware, so why not just modify firmware to forward the command of the fan to both?

nicedevil007, 2021-08-05:
It is already enabled in the firmware to react to M106 Pn Sxxx by manufacturer.

I already took a look at the BiquBX firmware they provided on their page and there are so much changes they made to stock marlin that I gave up.

Unfortunately the chip isn't compatible with Klipper right now. How Support will come in the future :/

supermerill, 2021-08-05:
I'm currently working on a new gcode writer, and you should be able to choose what is the gcode for a fan, so you may be able to set both. It's a medium-term goal.

nicedevil007, 2021-08-05:
Nice to hear that! I'm ready to test if it is implemented at any time :)

salocain, 2021-10-01:
if you have marlin inside you can modify this ,like this ( work on my btt octopus ) fan0 and fan1 works together

/**
 * Use one of the PWM fans as a redundant part-cooling fan
 */
#define REDUNDANT_PART_COOLING_FAN 1  // Index of the fan to sync with FAN 0.

CCS86, 2023-01-28:
I just added an auxiliary fan and would  love additional fan support.

I think the simplest implementation is to have it as a filament based setting. You just give it the fan #, the speed, and disable first X layers. 

RRF is pretty easy for fan control of multiple fans. Just change the P number in the M106.

Cracoukas, 2023-10-20:
... In my case I have a Biqu BX that got 2 ports for fans at the daughterboard sitting on the backside of the extruder. 1 Fan is not enough to get good cooling done here, so I installed a 2nd one at the front. Then I was able to enable 2nd fan manually with the gui of the display... 

Hi,
If you are still connected to this post I would like to know how did you actually activate the FAN 1 port of the SKR SE-BX board ? For me it does not react to  M106 P1 S255
Actually what I would realy like to know is where, on the main board,  is the power supply for the CNS-FAN.