supermerill/SuperSlicer #1452: The wall in the space for the cage nut is smaller than the outer perimeter

slimfy1, 2021-08-03:
**Describe the bug**
When the model is printed, the outlines of the place in the wall for the cage nut are visible. In Cura 4.10 i don't see that problem.

**To Reproduce**
Steps to reproduce the behavior:
1.Thin Wall -> Enabled
2. Gaps -> Enabled 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**

[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 
 - Version 2.3.55.7

supermerill, 2021-08-05:
You can try with 'overlapping extruder' -> 'all'
You can also reduce the gap fill overlap to 0 (it's in tab `print` -> `flow` )
Please report your results.

slimfy1, 2021-08-05:
@supermerill Hello!

Yes, that help me, thank you!
[image]

closed 2021-08-06 by slimfy1