supermerill/SuperSlicer #1456: Arc (G2/G3) support in gcode preview

JohanAR, 2021-08-05:
**Is your feature request related to a problem? Please describe.**
If one opens a gcode file containing arc moves then the preview becomes distorted

**Describe the solution you'd like**
Convert G2/G3 to a series of lines in order to approximately display them *in gcode preview* (i.e. this is not a request for arc support when slicing)

**Describe how it would work**
See above

**Describe alternatives you've considered**
Full arc support in SuperSlicer? :smile: 

**Additional context**
Not very high prio, mostly needed to help debug output from other programs, e.g. ArcWelder

supermerill, 2021-08-21:
do you have a gcode file to have a real test?

JohanAR, 2021-08-22:
This is the file I was given when I were trying to help someone debug their print

[gcode_with_arcs.zip](https://github.com/supermerill/SuperSlicer/files/7026889/gcode_with_arcs.zip)

bttguy, 2021-08-25:
Wow, you did it! Amazing @supermerill , huge thanks!

ETE-Design, 2021-12-16:
@supermerill Now that G2/G3 Preview is working, would it somehow be possible to show G2/G3 even before sliced, so you know the result? Have already ArcWelder in my Post-Processing but would be great to know the result even before Slicing...

supermerill, 2021-12-16:
Not possible. It can't modify the gcode before it's created.
Or did i misunderstood?

ETE-Design, 2021-12-16:
@supermerill Was thinking about, if I open an STL in SuperSLicer, then go to Preview then it won't show how it will look if I use G2/G3 Post-Processing... So have to "Slice" it first so it run my Post-Processing, then open my G-Code to se it... Or an I getting something wrong?

supermerill, 2021-12-16:
Oh, you mean using the gcode file after the post-processing in the gcode preview? yeah, I can add a setting for that.