supermerill/SuperSlicer #1459: Reload 3MF Bug/Question

randellhodges, 2021-08-06:
Not sure if this is a bug or by design.

If I export a mesh from Fusion 360 as a 3MF, I can load it into SuperSlicer just fine. (Well, it does sometimes place it off the build plate but I assume that is because the 3MF has the position relative to the center due to it being stored by Fusion 360). If I right click on the model, the option to reload it from disk is disabled.

Could SuperSlicer allow reloading from the disk when the source is a 3MF exported from Fusion 360? Maybe because there isn't a separate model file referenced by the 3MF is why it cannot be done?

supermerill, 2021-08-12:
A 3mf can contain many objects. Maybe thats why: you can't link an objet to a file. It can also contain settings.
I'll look if I can make an exception if there's only one object in the .3mf and no settings
Ideally, it should open a dialog to ask what to reload (settings, objets, etc..)

balloni55, 2021-09-20:
i have the same problem, the .3mf file createt with F360 is always positioned at the corner when importing.
The same file is positioned in the middle when importing with prusaslicer or S3D. 
The same body as stl is always positioned in the middle with SS
[Körper1.zip](https://github.com/supermerill/SuperSlicer/files/7197526/Korper1.zip)
[image]

supermerill, 2021-09-27:
> the .3mf file createt with F360 is always positioned at the corner when importing.

That's because the object is centered around the oringin, so it's placed around the origin.
If you want an auto-centered part, check `Configuration menu -> preferences -> auto-center parts`

balloni55, 2021-09-27:
Thanks for your helpfull answer, auto-center parts work fine

neophyl, 2021-09-27:
Also this is an issue over on the Prusa GitHub, apparently it’s fixed in their 2.4 alpha release https://github.com/prusa3d/PrusaSlicer/issues/6987