supermerill/SuperSlicer #1460: Gap Fill width higher than perimeter Width

DontTakeItForGranted, 2021-08-06:
**Describe the bug**
The Gap fill that is produced has a way higher width than the perimeter width

**To Reproduce**
Steps to reproduce the behavior:
use a relatively thick extrusion width. (i used 1mm for external perimeters and 1.1 mm for perimeters)
a 3d benchy at original scale is used in the first and second picture.
**>> Project File <<**
*Please* upload a ZIP archive containing the project file used when the problem arise. Please export it just before the problem occur. Even if did nothing and there is no object, export it as it contains your current configuration.

**Expected behavior**
the Gap fill Width should be in any case smaller than the perimeter width.

**Screenshots**
the perimeter width is 1.1mm but the gap fill width in this picture is more like 2.1mm :
[image]
[image]

the to thick extrusion of gap fill also leads to a extrusion of gap fill in mid air where a bridge should be that supports a perimeter.  
(not an extra bug, gap fill, when everything is how it should be, dont need bridge support because it is hold by the perimeters around it) 
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Mac os 10.15.7 catalina

**Additional context**
Add any other context about the problem here.

neophyl, 2021-08-07:
https://github.com/supermerill/SuperSlicer/discussions/1095

DontTakeItForGranted, 2021-08-07:
yeah that fixed it, 

The overlapping perimeter setting works but will lead to over extrusion 
so i dont understand why a perimeter and a gap fill shouldnt work where there is enough space for it.
I also tested it in  Prusa Slicer, PS seems to have the overlapping for all perimeters enabled, so i guess its fine

closed 2021-08-07 by DontTakeItForGranted

supermerill, 2021-08-07:
> i dont understand why a perimeter and a gap fill shouldnt work where there is enough space for it
> (maybe it would need a to big change in the slicing algorithm).

It's more difficult, as you can't do a loop. So you have to create a fake 'gap-fill-perimeter' a bit on the side, and then fill the rest with the normal gapfill.
Not easy to code.

DontTakeItForGranted, 2021-08-07:
thanks for the explanation.
i guess it would increase the overhang performance if the gap fill would be disabled if there are only overhanging perimeters around it  , but this could be to complicated to implement it(i dont know how slic3r/PS/SS is working internally)

[image]
[image]
gap fill is printed before overhanging perimeter:
[image]