supermerill/SuperSlicer #1461: Perimeter Seam Dive

CCS86, 2021-08-08:
I would love to see a feature that would insert a travel move, inwards, at the end of the external perimeter loop. You could set it as a "number of walls" to move inwards, so it tracked with perimeter spacing. Or, you could set as a distance in mm. The direction could be normal to the last print segment.

The thought is similar to wipe while retracting, but different. Rather than moving along a perimeter wall (and likely damaging it with nozzle pressure bleed), you move away from the external seam as fast as possible, before starting your retraction / z-hop. Especially on bowden setups, this will help make seams more crisp, and avoid creating artifacts during longer/slower retractions, or tall z-hops.

qwewer0, 2021-08-08:
I support this feature.

supermerill, 2021-08-08:
currently, the wipe is moving half a perimeter inward.
Problem is, if you want to go more "inside", if you do that where the part is thin and has only 1 perimeter-loop, you will destroy the other external perimeter.
So it's not that easy, as it has to check how far it can go...

CCS86, 2021-08-08:
I don't think it has to be that complicated, at least to begin with. If you are dealing with a part that has thin walls, you can set it to 0 or 0.5 walls, and forgo the feature.

supermerill, 2021-08-08:
The wipe already go 0.5.

CCS86, 2021-08-08:
I'm saying that if you have an edge case print, like one which has a total thickness of say 0.8mm right where the seam is, you could disable the seam dive feature.

In all other cases (the vast majority), you could run a value which gets the nozzle further from the external wall before retracting/hopping.

closed 2024-02-11 by supermerill