supermerill/SuperSlicer #1462: Bridge flow calibration doesn't work if layers are not set to 0.2mm

NameOfTheDragon, 2021-08-08:
### Version
2.3.56.7

### Operating system type + version
Windows 10

### Behavior
When performing "bridge flow calibration" the models do not print correctly if, say, the first layer is set to 0.3 mm.
This calibration screen also has no notes about interpretation of the results.

I like to print my first layer at 0.3 mm because it gives me more consistent results, however if I use that setting then the top layer of the bridge flow models is missing after printing (there is no number and the ironing infill doesn't seem to print or at least doesn't work well.

supermerill, 2021-08-08:
> When performing "bridge flow calibration" the models do not print correctly if, say, the first layer is set to 0.3 mm.

What is exactly the problem? Can you post your project file?

NameOfTheDragon, 2021-10-06:
There's no project file, it's the bridge flow calibration objects that are generated under the Calibration menu. Just set your first layer height to 0.3mm and then try to calibrate bridge flow. You will see the issue (the top layer will not print). I think it is because the model relies on 0.2mm layers. The fix is probably to force the layer heights to 0.2mm  in the object settings.

supermerill, 2021-10-06:
oh, ok.
it only happens if the first layer is exactly at 0.3mm, should be easy to solve.

> The fix is probably to force the layer heights to 0.2mm in the object settings.

it's already the case

closed 2021-11-07 by supermerill