supermerill/SuperSlicer #1463: [FR] Separate Bridge Infill/Perimeter Overlap setting

DontTakeItForGranted, 2021-08-08:
**Is your feature request related to a problem? Please describe.**
A separate for the Bridge infill/perimeter overlap, normally this is included in the infill/Perimeter overlap setting,
but a different (higher) value for Bridge Infill/Perimeter overlap can be very helpful in many cases, especially if a bridge flow is applied to overhanging perimeters that are connected to Bridge Infill. 

i know that there was a kinda similar FR on the Prusaslicer Github (i didnt found anything about it on the Superslicer Github), i open this FR because maybe something has changed in the code over the years to make this possible to implement or that it could be considered in the possible bigger code rewrite in the distant future of Superslicer that was mentioned somewhere.

here is a example pic of the Bridge infill that is sagging:
[image]
have a nice day:)

qwewer0, 2021-08-08:
Could you explain it a bit more to what would do to the part/image above?

DontTakeItForGranted, 2021-08-08:
yeah sure. 
in the image you can see that the bridge is sagging. this is because there are perimeters around it that re overhangs,
and because the bridge flow rate is applied to the there is basically no area where the bridge infill touches the overhanging perimeter, in theory this could be mitigated by applying the normal flow to this perimeter, 
but in reality you could not extrude as much material in a bridge as in a thick perimeter because there is no material to squish onto (i tested it  and the results where even worse)