supermerill/SuperSlicer #1464: Overhang overlaps External Perimeter

tome9111991, 2021-08-08:
Hello
don't know how to describe it.
I have the problem that the overhang overlaps the external perimeter.

[image]

At first I thought it was a bug in the GUI, but when I printed it it looked very ugly in these places.
with cura the surface looked a lot better.

.3mf
[Giroid_vase.zip](https://github.com/supermerill/SuperSlicer/files/6951082/Giroid_vase.zip)

supermerill, 2021-08-08:
That's only changing the speed&fan. you can remove it by setting a higher number in perimeter->overhangs->bridgefan & speed, or by deactivating overhang detection with a "0".

tome9111991, 2021-08-08:
Thats true i can  overhang detection with a "0".

But i like the setting for speed & fan.
 Maybe is possible to get a "clen" overhang layer?

supermerill, 2021-08-08:
something like that?
[image]

tome9111991, 2021-08-08:
Yes thats looks great

qwewer0, 2021-09-08:
This is how the shared 3mf file looks for me on 2.3.57.0
[image]

closed 2022-03-22 by tome9111991