supermerill/SuperSlicer #1465: How do I turn off Internal bridge infill?

EmreDogann, 2021-08-09:
### Version
SuperSlicer v2.3.56.7

### Operating system type + version
Windows 10 Pro 21H1

Hey there,
I've been trying to use SuperSlicer for a couple of weeks now but I've been having some crazy top layer over-extrusion. I've calibrated my e-steps several times, double and triple checked my flow, and made sure my extruder and hotend were all functioning properly. So I have narrowed it down to a SuperSlicer issue. I've sliced and printed this STL using cura and it turned it fine with no over-extrusion.

I suspect the over-extrusion is coming from the internal bridge infill because when I slice my file and switch to Height view, I see that, while the rest of the print is set to 0.08 layer height, the internal bridge infill is printing at a whopping 0.4mm layer height! Could this be the cause of the over-extrusion on the top layer? I have been through all the settings and have not found how to completely turn it off. Is there a solution to either turn this bridge infill off or reduce its printing layer height?

[image]

Thanks!

#### Project File (.3MF) where problem occurs
[Over-Extrusion Issue.zip](https://github.com/supermerill/SuperSlicer/files/6955590/Over-Extrusion.Issue.zip)

supermerill, 2021-08-09:
you can create thinner bridges by decreasing `bridge flow`.
Currently, it's set to 100% in your setting, so it will extrude at 0.4mm width and so 0.4mm height.
If you want a 0.1mm width & 0.1mm height, then you have to extrude 16 times less (4²) volume. So if it's what you want, you have to set the `bridge flow` to 6.25%

The current biggest issue is when you use variable layer height, as bridge flow can't adapt.
In the next prusaslicer version, they added a switch to disable bridge flow to avoid this kind of issues.
Imo, it's better to adapt dynamically to the current layer height, and use the height as the base... but I need to make or get many test to see what's best.

EmreDogann, 2021-08-09:
Understood, will use your suggestion until the next major update. Thanks!

closed 2021-08-09 by EmreDogann