supermerill/SuperSlicer #1466: Automatic part numbers/labeling

ciro1138, 2021-08-09:
As 3d prints are becoming more and more used for production it would be very nice to automatically add serial numbers to parts. This will help solve issues with customer complaints as well as QC in house for those who are concerned with this. 

How I see it working is adding the text/serial number area like painting the seam or support area. Text can be added either manually or preferably as you make copies it increases the starting string by 1 value. Debossing would be adjustable dimensions for text line width and depth.

supermerill, 2021-08-09:
Currently, you can 
* download freecad
* register its installation path in the preferences (should be `C:\Program Files\FreeCAD 0.19` on windows)
* add an object, like a 30mm box
* open the scripting window (under generate menu) and select "insert" instead of "replace"
* enter 
```
cut("stamped")(
	cube(30,20,20),
	linear_extrude(1)(
		text("SuSI 2.3.56",3),
	).rotate(x=180).move(5,5,1),
),
```
And it's stamped!
To choose the position, it's a bit striky, [here is a tutorial on how to get the right position](https://github.com/supermerill/SuperSlicer/wiki/Splitting-an-object-with-pins-&-holes)

replace the "5,5" by the coordinate of the start of the text (bottom left)
replace the two "1" by the wanted stamp height
replace the '3' to change the text size
you can choose the font if needed

I don't have the time for a nice gui in the medium term. If someone want to do it, it'll merge it.

ciro1138, 2021-08-10:
@supermerill 

Thanks for sharing this I completely overlooked that function. Works for now!