supermerill/SuperSlicer #1467: Heavy stringing between print and wipe tower when ”no sparse layers” on

hagan66, 2021-08-09:
When printing mmu prints with ”no sparse layers” i get heavy stringing to and from the wipe tower. See pics.

Version 2.3.56.7.

In preview mode without ”no sparse layers” retractions looks right, but with ”no sparse” on the hang in mid air. I dont know if that is a clue or not. 

I will upload the prj file in a minute also when Im at the computer.

[image]
[image]
[image]

supermerill, 2021-12-04:
should be fixed in the last nightly, can you test?

hagan66, 2021-12-07:
Hi!

I loaded my project file and now the wipe tower looks right but there are now these gaps in the print. See attached screenshot. I didn't start a print.

/Robert

hagan66, 2021-12-07:
[image]

supermerill, 2021-12-07:
will be fixed in this next nightly.

hagan66, 2021-12-07:
Cool, I will test it again then :-)

R

hagan66, 2021-12-08:
No gaps in the latest nightly build. 

However I get pretty sloppy supports since they vary from 0.1 (print layer height) up to 0.4 layer height. Is there a setting for this? I have looked/searched. I would like to limit support layer height to say, 0.2 mm for example. 

:-)

/Robert

supermerill, 2021-12-08:
you can reduce max layer height for the printer.

closed 2021-12-22 by supermerill