supermerill/SuperSlicer #1470: The speeds no longer work

tirasec, 2021-08-10:
### Version
2.3.57.0

### Operating system type + version
Windows 10

### Behavior
When you change the feed rates, they are no longer taken into account in the GCode nor in the simulation, everything is set to 8mm/s
[image]
[image]

#### Project File (.3MF) where problem occurs
[Crochet-SS2.3.57.zip](https://github.com/supermerill/SuperSlicer/files/6960465/Crochet-SS2.3.57.zip)

neophyl, 2021-08-10:
I downloaded the latest nightly which appears to be 2.3.57.0 and after I got it working (as theres a couple of missing bitmaps (resin_cog and sla_printer_cog) I opened your saved project file.

After slicing I saw the same as you with the speed set to a maximum of 8mm/s.  However on the loaded projects filament settings>filament>Print Speed Override the Max Volumetric speed is set to 8mm/s.  Which explains the limited speed.  Its set there.
Now I don't know if it should be set to that or not as its a PEI profile from the name and filament type settings.  I suspect the max volumetric of 8mm3/2 is being wrongly applied to maximum speed as the setting above that one which is set to 0 is Max speed.

tirasec, 2021-08-10:
Yes, you're right, the Max volumetric speed is applied to the max speed... I made a test with Max Vol. Speed to 50 mm3/s and the tool-paths speed are correct.
[image]

(For PEI, we use Max volumetric speed to ~8mm3/sec)

closed 2021-10-08 by supermerill