supermerill/SuperSlicer #1471: Is it possible to setup Wipe speed ?

tirasec, 2021-08-10:
**Describe the bug**
Is there a way to setup the Wipe Speed ? Which value is used for the wipe speed ?
With some strong materials we would like to reduce the wipe speed to a value close to the print speed.

**To Reproduce**
Steps to reproduce the behavior:
NA

**>> Project File <<**
NA

**Expected behavior**

**Screenshots**

**Desktop (please complete the following information):**
Windows 10
2.3.57.0

**Additional context**
Add any other context about the problem here.

supermerill, 2021-08-11:
> Which value is used for the wipe speed ?

travel speed

> Is there a way to setup the Wipe Speed ?

no yet.

tirasec, 2021-08-12:
It is a bit strange, the travel speed influence the lenght of wipe and looks to be 80% of Travel Speed... 

With a travel speed of 200 mm/sec
The Wipe Speed is 9600 (160mm/sec)
The wipe length is like that
[image]

With a travel speed of 100 mm/sec
The Wipe Speed is 4800 (80mm/sec)
The wipe length is like that
[image]

With a travel speed of 50 mm/sec
The Wipe Speed is 2400 (40mm/sec)
The wipe length is like that
[image]

It would be good to have the possibility to setup the wipe length and wipe speed...

[WipeLength.ZIP](https://github.com/supermerill/SuperSlicer/files/6973748/WipeLength.ZIP)

supermerill, 2021-08-12:
Wipe move when you retract. So the time depends of the retraction length and the retraction speed.
So the distance moved depends of the travel speed.

closed 2021-10-08 by supermerill