supermerill/SuperSlicer #1474: Default value for Max layer height prevents the slicing.

liftbag, 2021-08-11:
**Describe the bug**
If max layer height limit is set to 0 slicing is not enabled.

**To Reproduce**
Steps to reproduce the behavior:
1. Set Max layer limit to default value (0)
2. Import any stl file
3. The following error message is shown: First layer height can't be greater than max layer

**>> Project File <<**
[45_degree_modifier2.3mf.zip](https://github.com/supermerill/SuperSlicer/files/6968512/45_degree_modifier2.3mf.zip)

**Expected behavior**
The default value 0 should be tolerated and used as described in the tooltip

**Screenshots**
[image]
[image]

**SuperSlicer nightly macos 394**
**MacOS 11.2.2**

supermerill, 2021-08-11:
Oh right.

liftbag, 2021-08-21:
Solved with the 396 nightly build.

closed 2021-08-21 by supermerill