supermerill/SuperSlicer #1476: Can a script be implemented to control Neopixels on layer changes for print progress

BobCraig10, 2021-08-12:
A script to add M150 gcode to layer change that would light neopixels progressively or with color change to  show print progress, Heat and cooldown progress lighting could also be done in the start and end gcode. 

I could show a post process script I have for cura.

Thanks, Great work!!

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

supermerill, 2021-08-12:
you can write that in the custom after layer change gcode

something like
```
M150 R{layer_num==0 ? first_layer_temperature[current_extruder] : temperature[current_extruder]} G{layer_num / total_layer_count} B120
```

for the `cooldown`, i don't understand what's that.

BobCraig10, 2021-08-12:
> you can write that in the custom after layer change gcode
> 
> something like
> 
> ```
> M150 R{(layer_num==0 ? first_layer_temperature[current_extruder] : temperature[current_extruder])} G{layer_num / total_layer_count} B120
> ```
> 
> for the `cooldown`, i don't understand what's that.

That is all well above my pay grade, lol.   Can you by chance recommend something for me to read to start learning this? I was pretty proficient in basic programming 30 years ago, and have just started playing with Arduino.   I don't even knw what "language" that is, or if it is just a part of gcode.   
would I have to alter that, or put it directly in the " after layer change" area?

Thank you very much for answering!!

supermerill, 2021-08-12:
https://help.prusa3d.com/en/article/macros_1775

closed 2021-09-06 by supermerill