supermerill/SuperSlicer #1484: Perimeters Printed in Open Space for Overlapping Bridged Areas

BlueDragonX, 2021-08-13:
**Describe the bug**
Bridging does not behave as expected when slicing a model with bridged areas above overlapping shapes of different heights. Perimeters are printed around the smaller shape instead of bridging. This happens even when the smaller shape starts one layer above the larger.

**To Reproduce**
Steps to reproduce the behavior:
1. Overlap some shapes in negative space.
3. Slice it.
4. Evaluate the bridging.
5. Wonder how that'll print and yolo it.

**>> Project File <<**
[klicky_superslicer.zip](https://github.com/supermerill/SuperSlicer/files/6979821/klicky_superslicer.zip)

There are two files in this project. The first has a through hole from top to bottom. You can see it print the perimeters. The second has 0.2mm of material at the bottom of the smaller hole at the same height as the bridge. The perimeter is still printed.

**Expected behavior**
It does not print perimeters within a bridged area.

**Screenshots**
The bottom of the two objects. You can see the perimeters printed within the circular areas.
[image]

This is the problem layer from the top. You can see on the top that these are perimeters.
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 2.3.56.8

BlueDragonX, 2021-08-13:
With a bit of fiddling I was able to improve this by setting "No perimeters on bridge area: Keep only bridges". No other value of this setting had any effect.
[image]

This resulted in a significant improvement with solid bridge across the entire area. However, the overlapping area has some sharp corners and overlapping lines.
[image]

Setting bridge angle improves it even further:
[image]

I suspect any further improvements will need to be made in the model.

I do have one remaining question: What effect should the other values of "No perimeters on bridge area" have?

neophyl, 2021-08-13:
I'll just leave you this link to the wiki page https://github.com/supermerill/SuperSlicer/wiki/No-perimeters-on-bridge-area

BlueDragonX, 2021-08-13:
Thanks. That's enlightening - or it would be if the docs matched the behavior I'm experiencing. It explicitly says not to use the option I've enabled: "Too experimental, please don't use it until it improves a bit." Based on the docs I would think "Fill the voids with bridges" would have fixed the problem here but it has no effect and neither did the other options.

closed 2021-09-06 by supermerill