supermerill/SuperSlicer #1488: ctb support after sdk release

linx25, 2021-08-14:
Hi 

it seems Elegoo and chitu will release a sdk for the *.ctb fileformat. It would be great superslicer will support this
 
source of the announcement:
https://www.facebook.com/ELEGOOOfficial

supermerill, 2021-08-18:
I don't have facebook. Where do they publish their sdk / ctb file specifications?

linx25, 2021-08-18:
it seems. 

[image]

supermerill, 2021-08-18:
I won't integrate proprietary software.
Imo, best way to move forward is to reverse-engineer the .ctb format.

rudetrooper, 2021-08-23:
Apparently these guys managed to reverse engineering the new .ctb format. https://github.com/CryptoCTB/LiberatingMarsCLI

aelnagger, 2023-04-27:
I briefly looked into this. I have a mars 2 and am interested in being able to stick to one slicer. There's a fairly comprehensive writeup at https://github.com/cbiffle/catibo/blob/master/doc/cbddlp-ctb.adoc.

I think this can be done similarly to how CWS is configurable output format for SLA. I'm comfortable making an attempt at adding the configuration enum value, and creating a new concrete SLAAbstractArchive implementation based on the linked writeup.

zgrozemag, 2023-04-27:
UVtools already does conversion by just reusing headers https://github.com/sn4k3/UVtools