supermerill/SuperSlicer #1490: FR: Open superslicer with command line maybe not working correctly

ldiegos, 2021-08-18:
**Is your feature request related to a problem? Please describe.**
I have 4 printers all connected with each octoprint and each of them with a printer profile in superslicer. I'm a little tired and boring about opening new instances of superslicer and have always to change the printer profile, because it maintains the last opened.

I have been testing the command line with the "--load" parameter but it seems to ignore me or something is not working.
I understand that if i execute the "--load" parameter with a existent printer profile it will open with that printer profile. But, instead of that, it open with a new printer profile, with a new filament profile with the same name and a new print profile with the same name.

I don't know if as i have the printer profiles attached with the "physical_printer" profiles this is not working as supposed intended. 

This is what happen when i test:
Path structure:
**\physical_printer\:**
- A10MTesting.ini
- AvispoNejoA10.ini -->Inside is the preset  name linked to: AvispoNejoA10_0.4 nozzle
- Ender3.ini --> Inside is the preset name linked to: Asun3Pro_0.4 nozzle
- E180.ini
**\printer\:**
- AvispoNejoA10_0.4 nozzle.ini
- Asun3Pro_0.4 nozzle.ini
- 
- 
- 

When i write superslicer.exe --load "<path>\printer\AvispoNejoA10_0.4 nozzle.ini" superslicer opened and overwrite my profile and create a new print profile and a new filament profile with the same name. And of course is not linked with the physical printer.

If i try to change the --load to the physical_printer: superslicer.exe --load "<path>\physical_printer\AvispoNejoA10.ini" Then superslicer creates a new printer profile on that route and overwrite the physical printer profile with a printer profile, and also create a new print profile and new filament profile.

**Describe the solution you'd like**
I'm wondering if SuperSlicer can handle some command line parameters as --load_printer --load_print and --load_filament or maybe only the "--load_printer/--load" but working without the full path or detects if the path is pointing to the physical printer or the printer.

Is it possible that when you use the --load first of all check the physical_printer path to search for the profile: "<path>\**physical_printer**\AvispoNejoA10.ini"? it will load the physical printer with the printer profile in it and do not create anything.

If not found in the physical_printer path, then go to the printer and search for it:
--load "<path>\**printer**\AvispoNejoA10.ini" if found then load it and do not create anything else.

If not found then yes, create all the stuff.

**Describe how it would work**
Already explained in the previous section.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

supermerill, 2021-08-20:
currently:
 - you open the slicer normally
 - you select your profiles
 - you export the config to "C:/profiles/my_profile.ini" (or somewhere else)
 - you start a new slicer instance with `--load "C:/profiles/my_profile.ini"`
 - the new slicer instance will open with the saved selection of profiles.
 From my little test, it works with all 4 settings category.

But his only works if we want to select this exact combination that won't ever change.
I'm going to do see if i can do something to be able to import "internal bundle"

ldiegos, 2021-08-20:
> I'm going to do see if i can do something to be able to import "internal bundle" (and importing bundle seems like it's kinda broken currently)

Yes please, so appreciate.

For me is enough to load the physical_printer, because the printer profile is inside, but it would be wonderful if physical printer(with printer profile) and "print settings" could be load at once, then just i only have to choose the filament and send to the printer.

supermerill, 2021-08-21:
In the next nightly:

you export a config bundle with physical printer.
- if you --load it, it will be loaded (and it erases the current presets if they have the same name, but without saving so everything should go back to normal on the next reload)
- if you keep only the `[presets]` section in the bundle ini file, it will load these presets if their names can be found.

ldiegos, 2021-08-22:
Hi Supermerill, 

> you export a config bundle with physical printer.

I will export only the printer(printer_settings tab), won't i? But i can't in 2.3.56 it export also the "print settings tab" (so lets wait to the nightly)

> 
> * if you --load it, it will be loaded (and it erases the current presets if they have the same name, but without saving so everything should go back to normal on the next reload)

Understand, but, if only if, import the printer settings tab with the physical printer, it could be an option, because you don't change often the printer settings... but if you change something, you will need to export again(and not forget). But 'better than a slap in the face'/'better than nothing'/menos da una piedra' :)

> * if you keep only the `[presets]` section in the bundle ini file, it will load these presets if their names can be found.

sorry, i don't understand: "`[presets]` section in the bundle ini file"
You mean the "preset_name = AvispoNejoA10_nozzle0.4" inside the physical_printer.ini? this should not change, at least if i continue using the 0.4nozzle :) 
or 
the "default_print_profile = 0.16mm NORMAL 0.4 nozzle" inside the printer.ini? this can't be changed directly from superslicer, you should edit the printer ini file, but no problem.

Thank you so much for your patience.
Luis.

supermerill, 2021-08-22:
if your ini contains only`
```
[presets]
print = 0.20mm SPEED @MINI
sla_print = - default -
sla_material = - default -
printer = Original Prusa MINI & MINI+
filament = Generic ABS @MINI
physical_printer = test2
```

then it will try to selected presets with these names.

if it also has
````
[print:0.05mm ULTRADETAIL 0.25 nozzle]
allow_empty_layers = 0
avoid_crossing_not_first_layer = 1
avoid_crossing_perimeters = 0
avoid_crossing_perimeters_max_detour = 0
bottom_fill_pattern = rectilinear
bottom_solid_layers = 10
bottom_solid_min_thickness = 0
bridge_acceleration = 300
bridge_angle = 0
bridge_flow_ratio = 0.7%
bridge_overlap = 100%
bridge_speed = 20

[presets]
print = 0.05mm ULTRADETAIL 0.25 nozzle
sla_print = - default -
sla_material = - default -
printer = Original Prusa MINI & MINI+
filament = Generic ABS @MINI
physical_printer = test2
```

then the '0.05mm ULTRADETAIL 0.25 nozzle' will be added (or updated if it already exists) before being selected