supermerill/SuperSlicer #1499: Large gaps in

laptop-laptop, 2021-08-23:
**Describe the bug**

Print made by 2.2.53 only had two very small holes in between the two big circular holes. Needed to put it in hand's distance to see those two small 
[image]
holes. The rest of the surface looked solid. However, when made under 2.3.56.6, not only those two very small holes became large gaps. Many more big gaps of different sizes between perimeters (both circular and linear) were created for no reason.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version [e.g. 2.3.55.5], Use `Help->About`_: 2.3.56.6

**Additional context**

Are there any settings that I should change to eliminate those gaps created in 2.3.56.6?

supermerill, 2021-08-23:
2.2.53 or 2.3.55 ?
I advise to update to latest stable (2.3.56.8).

Can you 

for these gaps, you have two solutions to hide them (beside using a smaller nozzle / width).
 * activate 'one perimeter on top' (with 0 min-width if on version 2.3.55 because it's buggy)
 * use ironing post-process

laptop-laptop, 2021-08-23:
Thank you. I used to use 2.2.53 but tried 2.3.55 today. Then, I have issues with big gaps.

I will install 2.3.56.8 and print using your suggestion. The gaps are on the first layer at the bottom. How to have more over extrusion only on this layer?

supermerill, 2021-08-23:
If it's a regression, i need your project file (File->Save As) used to create the gcode from the 2.2.53 and the 2.3.55 version, to check & fix it.

> How to have more over extrusion only on this layer?

Adjust your bel levelling to have elss space between it and the nozzle. Or increase the first layer flow.

laptop-laptop, 2021-08-23:
Thank you. How can I increase the first layer flow only? Not sure if doing Ironing on the bottom layer helps. Can it be done under SuperSlicer? Prusa Support said that it cannot be done. They just suggested lowering the extruder's height.

supermerill, 2021-08-23:
> How can I increase the first layer flow only

search for "first layer flow" in the search tool.

> Not sure if doing Ironing on the bottom layer helps

it does not, that's why i wrote 'assuming it's a top surface'

>

laptop-laptop, 2021-08-23:
Thank you. I am very sorry that I cannot post project file for the public to access as it is work related. If I could, I would do it. If I don't post, it means I cannot post it due to privacy. Hope you understand my situation.

supermerill, 2021-08-23:
sure.
If you have some time, you can clip the problematic region (~5 cuts) to avoid these kinds of problems.

closed 2021-09-06 by supermerill