supermerill/SuperSlicer #1500: Jerk Limits Using Incorrect Units for RepRapFirmware

bradvaughan, 2021-08-23:
### Version
2.3.56.8

### Operating system type + version
Windows 10 Pro

### Behavior
The "Jerk limits" in the "Machine limits" section on the "Printer Settings" tab indicates the units are in "mm/s".  However, RepRapFirmware uses mm/min for this setting.  This is an issue for both the time estimates as well as the "Safeguard" and "limit" options.

The resulting M566 command in the generate G-code simply copies the value from the screen, so it's off by a factor of 60x.  Thankfully, this is in the direction of being slower than intended, but a bug nonetheless.

Additionally, this causes issues for the time estimates, because you need to put in the correct mm/min values to get the appropriate G-code, so the time estimates end up artificially low.  Again, this isn't a huge issue given jerk's relatively minor impact on overall print time in comparison to feed rates and acceleration.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
N/A.  It occurs on all G-Code exports for RepRapFirmware that I've tried.

supermerill, 2021-08-23:
I though I made it, but maybe I forgot to merge...
#632 #440

supermerill, 2021-08-23:
> Additionally, this causes issues for the time estimates, because you need to put in the correct mm/min values to get the appropriate G-code, so the time estimates end up artificially low. Again, [...]

That shouldn't happens, it's converted from mm/min to mm/sec. Maybe it's the feedrate unit that is creating some problems...
Can you post your project where the time estimate is problematicaly slow ?

bradvaughan, 2021-08-23:
Will do this evening. I appreciate the responsiveness.

bradvaughan, 2021-08-24:
For me, the values are copied directly and not being converted from mm/s to
mm/min.

[image: image.png]

[image: image.png]

supermerill, 2021-08-24:
I don't see any image and I can't download your project file.

bradvaughan, 2021-08-24:
I tried to do it via email. I'll do it directly on GitHub tonight.

bradvaughan, 2021-08-24:
[image]

`M566 X1200.00 Y1200.00 Z1200.00 E40.00 ; sets the jerk limits, mm/sec`

[3DBenchy-jerk-test.zip](https://github.com/supermerill/SuperSlicer/files/7042608/3DBenchy-jerk-test.zip)