supermerill/SuperSlicer #1502: Supports seems to be completly broken?

nicedevil007, 2021-08-23:
**Describe the bug**
My support seems to be completly broken with Superslicer. Everything works as expected if I use Ideamaker f.e.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. I used those Support settings with 0.16 layer height and 0.2 layer height
[image]

2. I also tryed these with 0.12 layer height
[image]

**>> Project File <<**
the problem arise on every print with support, 100% reproduceable.

**Expected behavior**
Support should print like it should

**Screenshots**
posted above

**Desktop (please complete the following information):**
 - OS: Windows 11 22000.160
 - Version 2.3.56

**Additional context**
I didn't want to add a new issue and tryed my best in discord and friends using superslicer, but we couldn't figure it out :( Hope anyone here can help me.

supermerill, 2021-08-23:
`>> Project File <<`

your project file?

nicedevil007, 2021-08-24:
[projectfile.zip](https://github.com/supermerill/SuperSlicer/files/7041974/projectfile.zip)

supermerill, 2021-08-24:
you set a max layer height of 1 for a 0.4 nozzle, that can't work well.
You sohouldn't put more of 80% of your nozzle height as max height.

I have to add a warning for that...

nicedevil007, 2021-08-25:
Oh man! Thank you very much, 3 guys looked over my settings and nobody found the issue. will test soon, and thank you for the warning included in a future release!

commit 30184cc0 (2019-01-04): Fix #1502 Upload to duet does not start the print
Code was erroneously removed during refactoring
closed 2021-11-07 by supermerill