supermerill/SuperSlicer #1510: [FR] Wipe on infill, inside perimeters

thiccnugget, 2021-08-28:
Superslicer 2.3.56
Windows 10 1909

Feature request:
move nozzle wipe on infill

I really love the "wipe" function, but it always triggers on perimeters, causing ugly blobs and curling.
So I was thinking about a box option to enable wiping on the infill to preserve print quality and to dicrease oozing and (possibly) retraction distance.

supermerill, 2021-08-29:
duplicate of #1461, i guess?

thiccnugget, 2021-08-29:
Well, not really, but similar. 
What I mean is to put a checkbox in retraction settings to enable wiping on the infill (only if present) and not on perimeters.

supermerill, 2021-08-29:
if i implement the 1461 "wipe dive" feature, and if someone set the (max) distance higher than the # of perimeters, isn't it similar to a wipe into infill?

thiccnugget, 2021-08-30:
Absolutely, that would be great!