supermerill/SuperSlicer #1514: Too many perimeters around a bridging infill causing the bridge to not overlap and connect properly

simpat1zq, 2021-08-29:
**Describe the bug**
I have a single model that is made of 5 non-touching pieces. 1 of those pieces is almost identical with another one. The only difference is that one has the word "good" embossed in it, and the other has "Evil" embossed in it. For some reason, one of those have perfect bridging, but the other one draws too many perimeters on that bridging layer, and causes the bridging overlap to be super short, and so the bridged infill doesn't connect for a bunch of the bridged lines. I've tried using the latest release version and in a recent nightly build(but not the latest). 

[image]

Here's the project file. 
[TestTray.zip](https://github.com/supermerill/SuperSlicer/files/7073225/TestTray.zip)

simpat1zq, 2021-08-29:
So it looks like the unchecking "Extra Perimeters: On Overhangs:" fixes that problem, but there seems to be a bug in that feature. If I rotate it 180 degrees, the problem is still on the piece that is on the near side of the build plate. If I rotate it 90 or 180 degrees, the problem is on the same side of the build plate. So that seems that it's definitely not a problem with the model, but a problem with that feature that causes it to create too many internal perimeters on the second one.

supermerill, 2021-08-30:
maybe a similar error as #1157

closed 2021-11-07 by supermerill