supermerill/SuperSlicer #1515: Layer Time Goal is not respected - feature print speeds are not reduced at all

gpread, 2021-08-29:
2.3.56.8

Creality CR-10, Marlin 2.0.8.2

The layer time view shows some layers with a  very short time (6 sec), when its expected to be 10 secs or more.

Those layers are actually purely auto generated support material layers which are within the object which itself is quite small (30x20x15mm).

Hence without any speed reduction (Support material speed is at 30mm/s) its to be expected that the time of such layers would be very short.

[image]

The Layer Time Goal is set to 10secs, Max Speed Reduction set to 90% and Min Print Speed set to 15mm/s.  I have also tried bumping up the Layer Time Goal to much higher values (e.g. 100 secs) and there is no change in the outcome at all. All the layer times shown stay the same.

[image]

As can be seen in the Speed view the speed of all the support material for all layers is the same 30mm/s (the current standard setting).  If the goal was being respected in this case, then just those particular support material layers would have the speed cut down to at least 20mm/s

I am printing PETG and trying to use the FAN for other than just bridging/overhangs causes wider issues (exacerbated by the fact that currently the speedup delay/advance for FAN has a bug too, see https://github.com/supermerill/SuperSlicer/issues/939#issuecomment-907836209)

So currently the resulting overheating on those 6 sec support material layers is causing subsequent adjacent layers of the object itself to get messed up.

I can't find an elegant/easy workaround either since a modifier does not allow adjustment of the speeds of support material or apply a specific FAN speed at all ?

In this somewhat unique scenario I can clobber the support material print speed globally though it still adds a lot of time to the print of each object  (~+7%).  For other more complex prints that solution is totally unworkable due to the time penalty incurred, meaning the only good resolution would be to post process/modify the output gcode by hand.

supermerill, 2021-08-30:
I need your project file to reproduce.

gpread, 2021-08-30:
Here you go, sorry should have done that right off the bat!

After more efforts to get my print better my sense is that the Layer time goal setting sometimes is having an influence but its not predictable or as expected and currently I am unable to give you additional concrete clues that helps identify further when/why/how it doesn't.

Hence I was tweaking, and I now don't have the project anymore with the exact same settings when I made the screenshots. 

However, the one below has slightly different settings but shows the exact same issue though (Goal of 30secs, 90% & min 1mm/s, supports speed@30mm/s - results in still some layers with 9sec and 25sec times and all supports shown with speed of 30mm/s).

[SSlicr_Bracket for 6mm SQ Arm_SSlicr_Issue#1515.zip](https://github.com/supermerill/SuperSlicer/files/7077072/SSlicr_Bracket.for.6mm.SQ.Arm_SSlicr_Issue.1515.zip)

Good luck and thanks in advance!

supermerill, 2021-08-30:
I loaded your project file and i have that:
[image]

there is no object layer with time below 30sec.
Also, the algo takes into account the time it takes to print the supports, so if you use more than 30 sec for support +object layer, then it won't trigger

gpread, 2021-08-31:
Ah ok, so perhaps it was just a misunderstanding (mis-expectation) on my part.  I did say in my initial report that the layers where I was seeing the goal as not being respected were purely support layers (no object extrusions at all were occurring on them).

So yes  now understood and accepted, the algo considers supports in that it won't slow down the object extrusions if the support extrusions take the total layer time to be >= goal time.  However, at the same time, if there is no object extrusion occurring then it allows those layer times to be less than the goal time. 

Thanks for clarifying, though I believe this behaviour can make for problems with melty supports (which may then contact the object in later layers), since the next layer can end up printing on top of the previous support only layer very quickly (in my case within a few seconds).

supermerill, 2021-08-31:
> However, at the same time, if there is no object extrusion occurring then it allows those layer times to be less than the goal time.

it uses object layer + support layer, to compute the time. So if there is no object layer at a certain height, with too little support, it should slow down the support print.

gpread, 2021-08-31:
> I loaded your project file and i have that:
> [image]
> 
> there is no object layer with time below 30sec.
> Also, the algo takes into account the time it takes to print the supports, so if you use more than 30 sec for support +object layer, then it won't trigger

So surely then there should not be those layers shown in shades of blue and green (which represent  22, 17, 12 & 9 secs layers)  ?
And/or if you look at the speed view the supports for those layers/areas of the print they should be shown as having been adjusted down from their normal of 30mm/s ? (but they are not). 

Or does it slow such supports only layers down by adding a dwell or delay in some other way ? 

I believe I also found another clue/scenario where the speed/time is not adjusted as might be expected, it could be atomic or it might overlap and explain why (to me at least) its not currently doing what you are saying it should.

I have been testing/adjusting the settings for "small perimeter speed" and had been using 15mm/s.

I found that the Layer times  don't change towards the goal time - as the speeds are not being changed for small perimeters to less than what is set there, even if the Goal speed allows a slower speed setting (e.g. 1mm/s and 90% ).

[image]

[image]
 
[image]

[image]

[image]

[SSlicr_Bracket for 6mm SQ Arm_SSlicr_Issue#1515 - Small Perims.zip](https://github.com/supermerill/SuperSlicer/files/7085314/SSlicr_Bracket.for.6mm.SQ.Arm_SSlicr_Issue.1515.-.Small.Perims.zip)

commit 8bc8dfab (2019-01-25): Build: Fix install targets, add SLIC3R_FHS, check resource dir on GUI_App init
Fix #1515
closed 2021-09-06 by supermerill