supermerill/SuperSlicer #1517: Gcode file generated from STL is missing small bolt-hole features

holohed, 2021-08-30:
### Version
_Version of SuperSlicer used goes here (help->about)_
2.3.56.5

### Operating system type + version
_What OS are you using, and state any version #s_
Macintosh 10.13.6 

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_
  * _Screenshots if useful_
Possible bug?  

GCODE created by STL file import into SuperSlicer has a few features missing (STL had 6 bolt holes, GCODE only has 3 bolt holes).  

The same STL file imported into "Slic3R" does not have the same missing bolt holes.  

_Is this a new feature request?_
No

Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_
[2MF_JPEG_and_STL_files_Archive.zip](https://github.com/supermerill/SuperSlicer/files/7075223/2MF_JPEG_and_STL_files_Archive.zip)
ZIP file contains:
1. screenshot of STL file upon input (6 small bolt holes visible) 
2. screenshot of GCODE rendering (only 3 small bolt holes evident) 
3. 3MF project file
4. STL file that caused the problem.

neophyl, 2021-08-30:
Ok a few things.  First of all your object has errors.  SuSi is indicating this with the warning triangle.  Fundamentally this is why you are having slicing issues.
Secondly if I open your saved 3mf file and slice with no changes ALL the holes are there.  If I also import your stl file of the same part into your project and slice again the imported part is sliced with the missing holes.  I have seen this before, the act of saving to 3mf can in some cases in effect 'fix' a model.  I'm not sure why this is but the 3mf format is a lot better than the stl format for the storage of information.  

I loaded your model into blender to take a look at and there many many errors.  After using Blenders 3dprint tool to repair it (down to 2 small errors, zero faces) I loaded it into Super Slicer and sliced it.  The fixed model was sliced correctly too.

Slic3r has diverged quite a lot from PS/Susi and their model handling is different.  PS doesn't tend to handle slightly broken models as well as say Cura does for example.  PS2.4 when its released is supposed to handle some of the more common errors (such as you commonly get from OpenScad models) better.  I'm sure those changes will trickle down to future versions of Super Slicer too in the future.  
Until then you might want to make sure your models don't contain any errors or at least learn how to fix them when they do.
[filter_v1_3M_noxious_fumes_neo.zip](https://github.com/supermerill/SuperSlicer/files/7075602/filter_v1_3M_noxious_fumes_neo.zip)

holohed, 2021-08-30:
OK, thanks for the tips.  I'm a bit new.  

I didn't understand that I had to save to .3MF, then reload the .3MF file, in order to see the error warnings.  

My previous workflow doesn't involve .3MF files -- I just imported the STL by dragging-and-dropping the STL onto the Superslicer GUI.  So I don't get to see any errors, and there may not be a warning triangle using this workflow route.  

I might have to be a bit more careful and use the save/reload .3MF file workflow in future.  The other thing that seems to work is the SuperSlicer "File/Repair STL File..." -- but sometimes I don't know about the STL errors....

The problematic STL file did come out of Openscad, but I wouldn't necessarily blame Openscad for it -- the original-original STL file that I imported into Openscad had errors in it, and I was using Openscad to remove the errors and augment the STL shape somewhat.  The original-original-STL same from thingiverse, which in the past I've found to be rife with bad STL files.... So sometimes I like to be warned in a log window or somewhere that the STL is not so good.  I thought I'd fixed the errors in Openscad....but apparently I hadn't.   My apologies.  

Thanks for your help.

supermerill, 2021-08-30:
> I didn't understand that I had to save to .3MF, then reload the .3MF file, in order to see the error warnings.

It's not really a wanted behaviour. I didn't really support malformed stl, I let prusa work on it.