supermerill/SuperSlicer #1518: Option to replace space character with Underscore automaticaly in gcode file name

supermerill, 2021-08-30:
see #1516
``
This is a QOL idea, and one that is more of a workaround that may benefit Klipper users.

It appears that Klipper does not handle spaces in filenames especially well and causes certain issues like errors downloading files or generating thumbnails in the frontend software that is commonly used. My firsthand experience is with Moonraker and Mainsail over Klipper, but Fluidd may also have this problem.

In any case, I wonder if it would be possible to provide "checkbox" option to "Replace Output Filename Spaces with "_"" or something similar. This makes sense in the "Printer Settings" tab as this is a "printer" (Klipper) issue rather than a "Print Settings/Output options" issue.

My current process is that I have renamed my print settings, filament settings and printer profiles to not have any spaces, and also made sure not to have any spaces in [printer settings/output filename format], but I still have to watch out for STLs that have spaces in their filenames.
``

neophyl, 2021-08-30:
Sounds like a useful addition.  How easy would it be to extend that concept to also replace special characters ?  One thing I have to be careful of with my setup using a flash air card is using things like % or other special characters. If I transfer a gcode file over with one it makes the entire directory non accessible (a limitation of webdav I believe).

supermerill, 2021-08-30:
I can create a text field where you write all forbidden characters to replace with '_'

neophyl, 2021-08-30:
Sounds viable 😊 Although I try and avoid regex as it hurts my poor brain lol.

yiannias, 2021-09-07:
So thankful for addressing this!  

I am trying to find the actual place where this is visible in the interface (I have tried the PC and Mac version of the build and can't seem to find it).  

I would expect it in Print Settings/Output Options/ in the "Output file" group under "Output filename format" or in Printer Settings/General or in Preferences/General, but I can not find it there or anywhere else.

supermerill, 2021-09-08:
> I would expect it in Print Settings/Output Options/ in the "Output file" group under "Output filename format" or in Printer Settings/General or in Preferences/General, but I can not find it there or anywhere else.

it's in printer -> general -> firmware

you try a search with 'illegal' or 'filename'

you need to be in expert mode

yiannias, 2021-09-08:
Looking at my Mac install, I am not seeing this:

Am on 2.3.56.9, expert mode.  Search for firmware or illegal also did not
bring this setting up.  (Please see attached screenshot)

My install method is simply copying the .app file to the application
directory.  Is there a settings file that needs to be reset maybe?

[image: Screen Shot 2021-09-08 at 9.02.25 AM.png]

supermerill, 2021-09-08:
It's only on the 2.3.57 branch, as the 2.3.56 one only get bugfixes for now.

yiannias, 2021-09-08:
Oh! I see!  Thank you, I will move on to this branch!

Kadah, 2021-11-26:
Just going to note this hear since the SEO for this feature leads.

To make it more obvious to yourself that the setting for this isn't undefined when set to just " ", use "[\s]". This is the regex match for whitespace, eg. space, tab, and newline.