supermerill/SuperSlicer #1519: Ideas for further Improvements - Please keep going, I love what I see so far, its great!! :-)

gpread, 2021-08-30:
I love this slicer so far :-)

I had been searching for various different things to help me diagnose and tune my printer, firmware and skills.  In many cases I could not find anything at all or of any reasonable quality.  Once I found this I discovered it had good tools/features which answered a significant number of the needs I had developed.

Also the layout/organisation for settings is soooo much better, more intuitive, tidy, etc than on prusaslicer

So firstly a big pat on the back for all the work so far!!! :-)

The below ideas are in no particular order:

How about the ability after having generated/exported  the gcode to be able to navigate, via a view by clicking on a feature, directly to the relevant place in the actual gcode ?   (This is something perhaps easiest best built into the g-code viewer since by definition its post g-code generation).  This could help with debugging, research and/or making fine tweaks to production print g-code.

Expand the "click to hide" concept in the Feature Type view of Gcode Preview, to other views, such that its easier to see exactly which layers/features are at a particular speed, fan speed, layer time, etc. Currently you have to scroll back and forth on the layer selector to ensure you don't miss where certain things are/are not happening.

With overhangs provide the ability to include config for an "advance length", that is - specify a mm distance prior to the overhang that the fan, bridge speed, bridge flow, etc settings will be applied to.  (this potentially could replace or enhance the fan speedup option as there is clearly some overlap).  Currently there is no good way I can see to adjust perimeter speeds in the region of overhangs using modifiers as they result in completely new perimeter paths being formed creating new seams and points of weakness, rather than just a new segment in the existing perimeter at the new speed.

Have a modifier which allows at least perimeter speeds to be adjusted in a region without creating new perimeter paths, so as to allow tuning to improve detail in specific parts of a print (not just the entire layer).

Have a modifier which allows all speeds in a region to be varied by a percentage, typically less than 100%, again to allow detail preservation (due to small or complex features which are not the majority of the print).

supermerill, 2021-08-30:
> How about the ability after having generated/exported the gcode to be able to navigate, via a view by clicking on a feature, directly to the relevant place in the actual gcode ? (This is something [...]

It may be relatively easy to write the current line number in the app foot bar. You navigate with the two sliders.

> Expand the "click to hide" concept in the Feature Type view of Gcode Preview, to other views, such that its easier to see exactly which layers/features are at a particular speed, fan speed, layer [...]

As most of the other things are continuous (not discrete), it need a special range slider widget, and a mechanism to store visually the current filters (and a cross to erase them). It's a very big task, as gui stuff is very time-consuming.

> With overhangs provide the ability to include config for an "advance length", that is - specify a mm distance prior to the overhang that the fan, bridge speed, bridge flow, etc settings will be [...]

I added (borked) fan delay because fan takes time to start and no firmware takes that into account (because of the short gcode cache window?). Speed should be taken care of from the deccel / accel value. The flow shouldn't increase in advance, as it may create a pressure build that will release at the bridge start.

> Have a modifier which allows at least perimeter speeds to be adjusted in a region without creating new perimeter paths, so as to allow tuning to improve detail in specific parts of a print (not just the entire layer).

yes. But it needs a new gocde writer system (which is currently worked on)

> Have a modifier which allows all speeds in a region to be varied by a percentage, typically less than 100%, again to allow detail preservation (due to small or complex features which are not the majority of the print).

why not. But after that, you may have a too slow speed for some features, so you'll need a min_speed_region.

I advise you to create a separate issue for each request, as it's much easier for me to sort them and mark them when they're done.

gpread, 2021-08-30:
Thanks for the feedback.

I am all for the quicker but effective solutions in general, since I am not a professional programmer (though I am a dangerous hacker, lol) and I don't know this code at all yet, I make high level suggestions with no idea of the relative difficulty.

Sure, I will consider your feedback and raise/add individual issues/requests.  As much as anything my intention this time was to publicly commend all the great work so far.

closed 2021-12-15 by supermerill