supermerill/SuperSlicer #1520: Setting infill and top&bottom to zero makes 3rd layer much slower than it should be.

yonkiman, 2021-08-30:
### Version
2.3.56.8

### Operating system type + version
Windows 10

### Behavior
Slicing a cube.  When I slice it without infill and without top and bottom layers, the 3rd and all subsequent layers print much slower than they should:

I set First Layer speed to 25mm/s, 1st layer gcode generates G1 F1500.  Great
I set Perimeter speed to 80mm/s, 2nd layer gcode generates G1 F4800.  So far so good.
Then for the 3rd and all subsequent layers, the gcode generates G1 F1186!!!!!

When I make top and bottom and infill non-zero, the 3rd and higher levels all print at F4800.

#### Project File (.3MF) where problem occurs
[SlowHollowBlock.zip](https://github.com/supermerill/SuperSlicer/files/7080020/SlowHollowBlock.zip)

supermerill, 2021-08-31:
you set your min layer time to 15 sec in the filament cooling.

yonkiman, 2021-08-31:
I also enabled the "I am an idiot" function.  :-/

Though shouldn't the min layer time apply to all layers, not just those that have cooling?

closed 2021-09-06 by supermerill