supermerill/SuperSlicer #1522: Now you see it, now you don't - Layers sometimes not being displayed (view is blank)

gpread, 2021-08-31:
2.3.56.8

Creality CR-10, Marlin 2.0.8.2
Windows 10 - 20H2 (19402.1165)

I filtered the views using the "Click to Hide" feature in Feature Type view.
Then I switched to the Speed view and enabled one layer mode. 
Then I began to look at specific layers to establish which feature was at what speed (specifically I was interested in speeds of perimeters that had been qualified as small).  So I began to move between layers using up/down keys.

When I scrolled upwards (increasing layers) I would see layers that conformed (matched the filter applied in Feature Type view) but then on occasion scrolling back down, a layer that had shown features appears empty of its features with just a Toolmarker.
It seems that for some reason it decides on scrolling back (down keypresses) for those layers to move the tool to the start of the layer (and hence nothing is shown without also adjusting the horizonal slider to again show the entire layer).

See screen recording below of it happening in practise (in particular watch for behaviour with layer #41):

https://user-images.githubusercontent.com/59426970/131545058-ce8072a6-d2fb-4dfe-adbf-75b9ee27a7b5.mp4

Here is the project file where it was happening (and shown in the screen recording above).
[SSlicr_Bracket for 6mm SQ Arm_SSlicr_Issue#1515 - Small Perims.zip](https://github.com/supermerill/SuperSlicer/files/7085613/SSlicr_Bracket.for.6mm.SQ.Arm_SSlicr_Issue.1515.-.Small.Perims.zip)

gpread, 2021-08-31:
You have to watch the screen capture (its only 40 secs( it clearly shows layer 41 displays when scrolling vertical slider up and then is empty when the slider is moved back down.

supermerill, 2021-08-31:
oh, ok

supermerill, 2021-09-26:
That's the extruder position that is set to the begin instead of the end of the layer, i'll se if I can fix that.

closed 2021-11-07 by supermerill