supermerill/SuperSlicer #1526: Crazy Internal Brim perimeters when supports overlay it (different behaviour to same with external Brim)

gpread, 2021-09-02:
2.3.56.8

Creality CR-10, Marlin 2.0.8.2
Windows 10 - 20H2 (19402.1165)

Unlike when supports overlap/overlay an external brim (where it routes the brim perimeters around the support structures) with an internal brim, its creating the brim perimeters as is they were cut by the support structures.

[image]

This means they are pretty much unprintable as segments of the brim perimeters are tiny in length.  It also surely means the adhesion of the object is lower since the adhesion of the brim has been lowered.

To me, it seems like for both internal and external brims it would be technically easier just to consider once the brim has been calculated that its part of the print object and have the supports then sit on top.  Currently it makes the brim's more complex and difficult to print, which defeats their objective somewhat (to improve adhesion of the 1st layer).

gpread, 2021-09-02:
Project file for the screenshot & Issue

[InsideBrim_with_Supports_SSlicr_Issue#1526.zip](https://github.com/supermerill/SuperSlicer/files/7099093/InsideBrim_with_Supports_SSlicr_Issue.1526.zip)

gpread, 2021-09-07:
2.3.57.0, Creality CR-10V3 (modified), Marlin 2.0.8.2 & Windows 10 - 20H2 (19402.1165)

Just tried this and its clearly a big improvement for the Internal Brim case, Thanks!

However, Brim and supports interaction (may apply to raft also, I didn't check) to me is still somewhat clumsy, wasteful, hard to print and its also inconsistent depending on whether the brim is internal or external.

It would not be such an issue if there was not  also often unnecessary support structure placed on the 1st layer.  When there is a feature needing support but which adjacent to but _**only**_ on one side of the perimeter/external wall of the object ,  there is a small support line created on the opposite side of the wall.  This cannot be prevented by using modifiers nor enforcers from my experience.

[image]

[image]

gpread, 2021-09-07:
So with internal brim it now looks like 

[image]

Which alone is usable/workable.  However, if external brim is used (alone or in addition) then it get corrupted and the interaction of support and brim(s) are ugly (it tries to tightly hug the support and causes the perimeters to have tiny but frequent perturbations)

[image]
Both Internal and External

[image]
Just External

My preference/suggestion would be the cleanest solution would be to have supports processed only after brim and/or raft is applied (so they sit on top and the brims are only clean/simple perimeters).  I believe this would also be the most robust/effective in terms of the purposes of  brim and supports and also allow the highest practical speeds for the first layer. 

If not that then its desirable to at least get the external brim behaviour to mirror the new/current internal brim behaviour for interaction with supports.

supermerill, 2021-09-08:
> My preference/suggestion would be the cleanest solution would be to have supports processed only after brim and/or raft is applied (so they sit on top and the brims are only clean/simple [...]

I'll see that after the merge with prusa 2.4, as they changed the support generation.

gpread, 2021-09-08:
Cool. Good to know, thanks for your efforts so far.

Supports are pretty poor currently I think it's fair to say.  Fingers crossed they have/are making some major improvements soon.

commit 237eaa5f (2019-01-04): Fix #1526 --datadir not working