supermerill/SuperSlicer #1528: Questions about ironing

laptop-laptop, 2021-09-05:
**Is your feature request related to a problem? Please describe.**

Hi, I don't quite understand what Infill->External patterns->Top and Bottom as well as
Solid pattern do. Could you please explain in more details?

If I choose Ironing under Infill->External patterns->Top, does that mean the top most layer will be ironed?
As for Infill->External patterns->Bottom, the description seems to suggest that choosing Ironing will 
iron the bottom layer.  However, the printed object did not have the bottom layer ironed.

As for Infill->External patterns->Solid pattern, what does it do if Ironing is chosen and coloured filament is used?

**Describe the solution you'd like**
A clear explanations on the above three parameters. Thank you

supermerill, 2021-09-05:
There is two ironing process to choose:
 - the infill one (it add a second pass over a rectilinear solid infill)
 - the post-process one (which extrude a bit more plastic on top of the object)

for your questions about the infill one:
- "If I choose Ironing under Infill->External patterns->Top, does that mean the top most layer will be ironed?" -> it will iron all surfaces that have air over them.
- "the printed object did not have the bottom layer ironed both in the visualization and in the actual print." -> it will iron the first layer, if there's no raft, because most of bottom layers are replaced by bridges after the first layer (unless if you're using soluble supports)
- "As for Infill->External patterns->Solid pattern, what does it do if Ironing is chosen?" it will iron all solid surface. the option is here for people that want to use it to test it to see if it's improving 100% fill clear parts. It's very niche. Note that the bottom ironing is also in this case, as it can't iron the visible part.
- " coloured filament is used?" coloured? why it is relevant?

laptop-laptop, 2021-09-05:
Thank you for your explanations.  I guess if I want to use the "infill one", I just turn on the ironing under Infill->Whatever. How to choose the "post-process one"?

Let's consider a C-shape part composed of two horizontal bars and one vertical bar.

- So  Infill->External patterns->Top means ironing only the top surface of both the top and bottom horizontal bars? 

- I don't use raft but I cannot see the bottom being ironed. By ironing the first layer, do you mean the top of the first layer? If that is the case, I certainly cannot see it. Under what situation is this used? In the C-shape example, do you mean it will just iron the top of the first layer of the bottom horizontal bar?

-   As for Infill->External patterns->Solid pattern ironing all solid surface, in the case of a cube, does it iron all six surfaces?

-  There is a short comment on "for Infill->External patterns->Solid pattern" mentioning about using it for clear parts so I wonder if it does anything for those who use non-clear filaments. Is there any advantage in using this option if I use for example white or black colors?

supermerill, 2021-09-06:
> How to choose the "post-process one"?

[image]

other questions:

difference between rectilinear and ironing:
rectilinear : [image]
ironing: [image]
(see the thin red stripes interlaced with purple ones)

surfaces:
[image]
only flat surface can be filled. sides are perimeters.
bottom are surfaces that doesn't have plastic just below, top are surfaces that don't have plastic on top.

> There is a short comment on "for Infill->External patterns->Solid pattern" mentioning about using it for clear parts so I wonder if it does anything for those who use non-clear filaments. Is there any advantage in using this option if I use for example white or black colors?

no, it just slows down everything and risks clogging your extruder.

laptop-laptop, 2021-09-06:
Thank you. For "thin red stripes interlaced with purple ones", in this case what do those thing red stripes represent?

If I use 2.2.53, how to choose between Infill ironing and post-processed ironing?

supermerill, 2021-09-06:
> Thank you. For "thin red stripes interlaced with purple ones", in this case what do those thing red stripes represent?

There is a fill done via a solid infill (purple) and then an ironing done with a very little flow over it (red, with top fill speed)

>So what does Infill->Bottom->Ironing do to the actual prints?

It does the same thing to the bottom surfaces.

laptop-laptop, 2021-09-06:
Thank you. Could you please let me know how to choose between Infill ironing and post-processed ironing in 2.2.53? My settings are tuned under this version and they don't work quite well under the latest version.

supermerill, 2021-09-06:
post-process ironing doesn't exist in 2.2.53 I think.

laptop-laptop, 2021-09-06:
Thank you. Is the post-process ironing worth to move to the latest version (2.3.56.8) even it means I have to do various tests again to find a new set of settings that will work well under the new version? 

How come even I set Infill->Bottom->Ironing to iron all bottom surfaces, the flat "Bridge infill" (overhang) layer just above the Support material interface is not ironed? I am talking about the bottom of the top horizontal bar of the C-shape object. The "Bridge infill" layer is made of lots of visible parallel lines. I want to smooth them out completely or make the lines less visible to make it more like a flat solid overhang surface.

supermerill, 2021-09-06:
> How come even I set Infill->Bottom->Ironing to iron all bottom surfaces, the flat "Bridge infill" (overhang) layer just above the Support material interface is not ironed?

> because most of bottom layers are replaced by bridges after the first layer (unless if you're using soluble supports)

you can't iron a bridge, because you can't push on it.

laptop-laptop, 2021-09-06:
Thank you for the explanation. If I don't use soluble supports, what is the best settings to use to make the overhang as solid nice looking as possible?

supermerill, 2021-09-06:
You can put some resin /paint on it afterwards...
You can also try with very low bridge flow.

laptop-laptop, 2021-09-07:
I tried 2.3.56.9 you released yesterday. Thank you for making such a nice slicer with lots of options.

By "very low bridge flow", how many percentage do you recommend? Currently I have With&Flow->Flow->Flow ratio->Bridges set to 75%.

Given that under "Ironing post-process", we can choose between "All top surfaces", "Topmost surface only" and "All solid surfaces", is there any point to choose Ironing under Infill->Solid, Top and Bottom? If not, is "Monotonic (filled)" the best option for Infill->Solid, Top and Bottom if I want nice looking top and bottom surfaces?

supermerill, 2021-09-08:
> By "very low bridge flow", how many percentage do you recommend? Currently I have With&Flow->Flow->Flow ratio->Bridges set to 75%.

You can go as low as 30% if the layer height isn't huge. This may decrease the quality of 'bridge overhangs' but increase a bit the bridging infill (bottom surface over supports). If you use support, you may need to reduce the gap between supports and the part.

> is there any point to choose Ironing under Infill->Solid, Top and Bottom?

This doesn't add flow, it doesn't over-extrude. So I don't recommend the post-process on "all solid" (unless you need to overextrude, like when you print with clear filament).

Other than that, thanks to the little overextruding, the post-process ironing should give you a better result in most cases.

> is "Monotonic (filled)" the best option for Infill->Solid, Top and Bottom if I want nice looking top and bottom surfaces?

for the bottom, the pattern will be visible in most case, so choose the pattern you want to see. Monotonic is better than rectilinear in this regard.
For the solid infill, unless you have a very specific geometry, choose the pattern that is the quickest as it shouldn't have any impact on surface quality (so rectilinear).

laptop-laptop, 2021-09-08:
Thank you. What about Infill->Top? Which option is recommended if I do a "Ironing post-process" with "All top surfaces"?

Is it possible to change the pattern of Bridge Infill, the layer just above the Support Material Interface? It is parallel lines by default.

supermerill, 2021-09-08:
> Which option is recommended if I do a "Ironing post-process" with "All top surfaces"?

monotonic

> Is it possible to change the pattern of Bridge Infill, the layer just above the Support Material Interface? It is parallel lines by default.

not possible, as only strait lines can be printed, as there is no anchors to do turns. (but the bringing algo isn't perfect, so its results can be suboptimal times to time)

laptop-laptop, 2021-09-08:
Thank you. 

I noticed that in 2.2.53, I could remove the support near the overhang witt some effort but not that difficult. However, in 2.3.56.8 and 2.3.55.8, using the same settings, it became very very difficult to remove the support below the overhang surface. So for each software upgrade, I may need to readjust some settings?

supermerill, 2021-09-08:
For overhangs, its old setting split for fan & flow. The default also by a bit, you can try to set both of them with the old value.

laptop-laptop, 2021-09-08:
Sorry when you mentioned to set "both of them", which two settings are you talking about? As for the old value, which setting's old value in 2.2.53 are you referring to?

supermerill, 2021-09-08:
overhang_width keep the flow control threshold as before. Now overhang_width_fan (or something like that) is the one that controls the overhang fan & speed (and so the colour).
Set overhang_width_fan to the value of overhang_width  to get back the old behaviour.

laptop-laptop, 2021-09-08:
Thank you. I searched but cannot find overhang_width_fan. Do you mean overhangs_width_speed? It already has the same value as overhang_width.

I doubled checked the values. In 2.2.53, Perimeters & Shell->Overhangs->On perimeters->'As bridge' threshold (overhangs_width) is 50%. In 2.3.56.8, Perimeters & Shell->Overhangs->threshold for->Bridge flow (overhangs_width) is also 50%. Same as Perimeters & Shell->Overhangs->threshold for->Bridge speed and fan (overhangs_width_speed) which is also 50% already. What else should I change?

supermerill, 2021-09-09:
The only other things that can change overhangs are the bridge flow, overhang speed and (filament) bridge fan speed.

closed 2021-12-15 by supermerill