supermerill/SuperSlicer #1529: Wipe Action applies "travel speed" to extruder REtract

Chris0812, 2021-09-05:
Hello, 
first the question is that a bug or just an inconvinient combination of my settings?
when I have in the Filament section under Overrides "wipe while retracting" activated, the maximum travel speed will be applied to the retraction in the GCODE: 

;WIPE_START
G1 F12000
G1 X171.424 Y171.778 E-0.76000

this is currently also the only setting checked in this section and wipe while retracting combination in the Extruder setting is disabled.

Before discovering this issue, I was using marlin, which apperently got a way of circumventing this in the Controlboard while printing....well, Klipper doesn't and throws a "MCU 'mcu' shutdown: Stepper too far in past". 

Easy to just leave that box unchecked but I still would like to adress this issue.

supermerill, 2021-09-05:
Can you post a project where the issue happens (and it does not)?
I tried with disabling extruder's "wipe while retracting"  and activating filament's "wipe while retracting" but it output the same gcode.

Chris0812, 2021-09-06:
Sorry, apparently I did mix up some things while trying the settings - correct cause would be "wipe while retracting" in the extruder section

Like you stated, the one in Filament section does work without issues

supermerill, 2021-09-06:
Wipe is made with travel speed. but the extruder speed should be good, as the speed is limited by the xy move and the extruder delta is here low.

The next version (2.3.57, tomorrow) will allow you to set a wipe speed.

closed 2021-10-08 by supermerill