supermerill/SuperSlicer #1532: [BR] 2.3.57.0 gone wild!!! the change to the Gcode preview is made whatever change you made

ldiegos, 2021-09-07:
**Describe the bug**
When you change something in the "print settings", "Filament settings", and "Printer settings" versión 2.3.57.0 which the "change to g-code preview" is active, automatically change to the g-code preview when you change one value... so you have to start jumping to the previous tab again and again. With other versions, jumping to the gcode tab was only made when you repositioned the object in the plate, but not with the other tabs.

If you change the "dropdown list" to "Don't switch" the Gcode-preview tab never update any more(losing the current sliced model) and can't press the button "slice" that appear in the previous versions. It has dissapear.

[image]

[image]

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Execute the version  2.3.57.0 over the 2.3.56.8 configuration files.
2. Click on preferences change the dropdown list to whatever option, expect "don't switch" and save.
3. Load a stl and try to change some configuration from the filament or printer settings.
4. See bug
5. Change the dropdown list to "don't switch".
6. Gcode preview tab never update again.
7. See bug.
8. 

**>> Project File <<**
*Please* upload a ZIP archive containing the project file used when the problem arise. Please export it just before the problem occur. Even if did nothing and there is no object, export it as it contains your current configuration.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Win10
 - Version 2.3.57.0

**Additional context**
Add any other context about the problem here.

RafaelSchridi, 2021-09-09:
Experiencing the same issue here, But if I set switch to preview to something else and I manually click on gcode preview while its slicing it will also not refresh.

ldiegos, 2021-09-09:
@supermerill  the problem is not just the "don't refresh" is that if you activate any other thing the superslicer starts jumping to the tab g-code preview what every configuration settings you change... That's horrible, annoying and boring.

In version 56.8 and 56.6 that not happen... the gcode is refreshed in background while you change the configuration but the screen not jump to it. only when change the position in the bed in the model tab.

locki-cz, 2022-03-03:
it is back in latest 2.3.57.11

CodeIsJustLikeMagic, 2022-03-09:
It is also in the 2.4.58.1 beta. Very annoying.

closed 2022-07-14 by supermerill