supermerill/SuperSlicer #1533: Don't refresh the 3D view unless really necessary 2.3.57

5axes, 2021-09-07:
Just tested the last 2.3.57.0 release. The new function "Don't refresh the 3D view unless really necessary"  doesn't work very well , even when I change a main parameter doesn't refresh the slicing preview.

Tested on Windows 10 Release 2.3.57.0 

Message : Slicing complete   I can export the G-Code but canno refresh the slicing result 
[image]

supermerill, 2021-09-07:
in preferences -> automation -> Switch to Preview when sliced, do you have "Don't switch"?

(see #1532)

5axes, 2021-09-07:
No : When the G-Code is ready  , but this settings doesn't have any influence . I'm also with background_processing = true but here alos dosn't have any influence if it's true or false

supermerill, 2021-09-07:
erf.
I'll have to change the notification system that refresh the data, it's a bit too spaghettis now.

closed 2021-11-07 by supermerill