supermerill/SuperSlicer #1535: Top Infill: "Ironing" ignores "Infill/perimeters overlap"

Pizzahd88, 2021-09-07:
**Describe the bug**
When setting Top Infill to "Ironing" the Infill/perimeters overlap value is ignored.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Print Settings -> Perimeters & Shell -> Infill: Top
2. Set it to Ironing
3. Slice an object
4. The infill does not overlap with the perimeters, only the ironing does. This results in small gaps along the edge

**>> Project File <<**
[Benchmark - All In One micro.zip](https://github.com/supermerill/SuperSlicer/files/7124323/Benchmark.-.All.In.One.micro.zip)

**Expected behavior**
The infill should overlap just as all other layers + ironing. This will help avoid the gaps along the edge of the ironing area.

**Screenshots**
Screenshot of the layer just below the topmost layer:
[image]

Screenshot of the topmost layer with Top Infill: Ironing
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 2.3.56

closed 2021-09-07 by Pizzahd88

Pizzahd88, 2021-09-07:
After much testing I have come to a conclusion that the current implementation results in great prints in some cases, and not so much in others.

If the infill is attached to the perimeter in an angle that is perpendicular +- 45 degrees (give or take), the current implementation does a great job with the top layer.

When the infill attaches to the perimeter with an angle closer to parallel it results in gaps along the edge as the holes between layer lines becomes greater due to them not overlapping with perimeter.

It seems like I have come upon the same print defect as: https://github.com/supermerill/SuperSlicer/issues/1290

Therefore I Close this issue and continue in that issue.

supermerill, 2021-09-08:
Just a note, I did that to ensure there isn't overextuding in thin area vs huge area. (or the reverse)