supermerill/SuperSlicer #1537: Disabled acceleration control still applies to print

nwgruber, 2021-09-09:
**Describe the bug**
When I disable acceleration control by setting the default to 0, the additional acceleration parameters (perimeters, travel, etc.) are still applied to the print if left nonzero. Seeing as these input fields are disabled after default acceleration is set to 0 I don't think this is intended behavior.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable acceleration control if turned off by setting default to some value
2. Assign a value to one of the other acceleration parameters (perimeters, infill, etc.)
3. Disable acceleration control by setting default to 0
4. Slice and print a model
5. Check the acceleration value on your printer

**Expected behavior**
When acceleration control is disabled and all the other fields are greyed out, I expected that they would not be applied to the print. In my screen shot below, the greyed out travel acceleration value would still be applied to the print.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 2.3.57
 - Using klipper gcode flavor

Laker87, 2021-09-09:
+

gpread, 2021-09-09:
Yes I have seen this also but....  I saw it with prusaslicer (I think 2.3.1), so I suspect this is an inherited bug. I didn't report it since it was easy enough to get around (setting all the other entries to zero before setting default to 0).

closed 2021-11-07 by supermerill